aircrack-ng/rtl8812au #120: Can't connect to AP while using wpa 2 Security

mrPink22, 2018-04-27:
i builded the driver sucessfully on Kernel 4.9.35-v7+ (raspberry pi) using hostapd Version: hostapd v2.7-devel-hostap_2_6-1555-g1fb0c10+ for Realtek rtl871xdrv.

when wpa 2 is enabled for the AP i can see the network but i can't connect to (with wpa=0 it workes fine) it with error message from hostapd:

hostapd_wpa_auth_disconnect: WPA authenticator requests disconnect: STA 5c:e0:c5:51:d5:d8 reason 2
wlan0: ap_sta_disconnect addr 5c:e0:c5:51:d5:d8 reason=2
nl80211: sta_remove -> DEL_STATION wlan0 5c:e0:c5:51:d5:d8 --> 0 (Success)

hostapd.conf: 

interface=wlan0
ssid=My_new_network
country_code=DE
hw_mode=a
channel=165
macaddr_acl=0
auth_algs=3
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=password
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
rsn_pairwise=CCMP
wpa_group_rekey=86400
ieee80211n=1
ieee80211d=1
ieee80211h=1
wme_enabled=1
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

do you guys have any idea how to fix that problem ?

kimocoder, 2018-05-18:
Have a possible fix for this in a week or something. Needs more testing before it gets pushed, so hold on a bit longer. Thanks

kimocoder, 2018-07-24:
We've disabled the regdom by adding a switch to the Makefile. It's disabled by standard, which should make hostapd work. Open another issue if the problem occurs again! Thanks

closed 2018-07-24 by kimocoder