aircrack-ng/rtl8812au #121: rtl8814au 2 AP simultaneously

ggerla, 2018-04-30:
is it possible to simultaneously create 2 AP at 2.4GHz and one at 5GHz?
when I bought AWUS1900, I thought so, but after numerous attempts, I started thinking that it was not possible. I have installed several drivers with different versions. But after installation I always see only one interface ...
is there anyone who managed this configuration?

ggerla, 2018-05-03:
looking at different posts on google I realized that with rtl8814au it is not possible to create two APs: one on 2.4GHz and one on 5GHz.
You can verify it with command iw list. At the end of message you can read "interface combinations are not supported".
There are a lot of chipset/driver that allow interface combinations, but only at the same frequency. So for example you can create an AP at 2.4GHz with WAP and another at the same frequency open.
I also understood that today there are no devices that allow to create 2 AP on different channel. Infact modern routers always integrate 2 chipset one for 2.4GHz and one for 5GHz.
Broadcom has announced a new SoC that will allow to solve the problem but at the moment there is no devices with this SoC.

I hope my last week's studies will also help others

closed 2018-05-03 by ggerla

kimocoder, 2018-05-03:
Great information provided here. thanks

kcdtv, 2018-05-04:
The problem here seems to be that "adding a virtual interface" does not work properly with our driver. It is basically the same issue than with airmon-ng.

>  Broadcom has announced a new SoC that will allow to solve the problem but at the moment there is no devices with this SoC.

It looks like qualcomm did it : [ipq4018](https://www.qualcomm.com/products/ipq4018)
You can find this SOC in , for example, the [ASUS_RT-AC58U](https://wikidevi.com/wiki/ASUS_RT-AC58U)
Saying so i have an alfa ac1200R that have a rtl8112au chip for ac and another realtek chipset for the b/g/n
The SOC used is called Panda, it is from realtek too. 
Maybe they used this configuration because the rtl8112au (and the rtl8114au) are not able to create simultaneously access points in both bands. 
What pieces me off is that this router works like a charm, i do not have any ping issue when i connect it to a 5ghz network while when i use any driver (this one or others) and conect my USB devices with this chip to a network, i have the ping that get very high every 30-45 seconds. Maybe it would inetristing to see what they have done with this router/SoC. Crazy shit...