aircrack-ng/rtl8812au #122: Can't unplug rtl8812au device or unload his module w/o reboot linux

alammer, 2018-05-03:
Hello! I'm sorry if I'm a little off topic. I build  rtl8812 au driver for Android Nougat 7.1.1 with a nethunter kernel patch and rootfs. I load the driver with the command "insmod /path_to_8812au.ko" in su shell. Sniffing in monitor mode, injection,  aircrack-ng works fine until the moment I disconnect the device from the USB port or try to down wireless interface with the command "ifconfig wlan1 down". In both cases, this causes the smartphone to reboot. In the logs it looks like:

[  278.864007] usb 1-1: USB disconnect, device number 3
[  278.865619] wl_cfg80211_netdev_notifier_call: handling state 9
[  278.865661] Unable to handle kernel paging request at virtual address 1400000014
[  278.865680] pgd = ffffffc00007d000

[  278.865701] [1400000014] *pgd=0000000000000000
[  278.865747] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[  278.865767] Modules linked in: 8812au(OF)
[  278.865816] CPU: 0 PID: 76 Comm: khubd Tainted: GF       W  O 3.10.73-nethunter-angler-2.0 #1
[  278.865841] task: ffffffc0c01bab00 ti: ffffffc00eb88000 task.ti: ffffffc00eb88000
[  278.865892] PC is at wl_cfg80211_netdev_notifier_call+0x228/0x298
[  278.865917] LR is at wl_cfg80211_netdev_notifier_call+0xa0/0x298
[  278.865937] pc : [<ffffffc00075536c>] lr : [<ffffffc0007551e4>] pstate: 60000105
[  278.865952] sp : ffffffc00eb8b930
[  278.865968] x29: ffffffc00eb8b930 x28: 0000000000000001 
[  278.865997] x27: ffffffc016278878 x26: ffffffc000ea90c8 
[  278.866067] x25: ffffffc091a2a800 x24: 0000000000000001 
[  278.866126] x23: 0000000000000009 x22: ffffffc001897000 
[  278.866154] x21: ffffffc024014000 x20: ffffffc000ea8000 
[  278.866183] x19: ffffffc033fd06a0 x18: ffffffc0819fae77 
[  278.866212] x17: 0000000000000000 x16: 0000000000000000 
[  278.866240] x15: 0000000000000000 x14: 2065746174732067 
[  278.866269] x13: 6e696c646e616820 x12: 3a6c6c61635f7265 
[  278.866299] x11: 696669746f6e5f76 x10: 656474656e5f3131 
[  278.866329] x9 : 3230386766635f6c x8 : 77205d3931363536 
[  278.866360] x7 : 382e38373220205b x6 : 0000000000000000 
[  278.866390] x5 : 0000000000000000 x4 : ffffffc0c1178a80 
[  278.866419] x3 : ffffffc0c117e088 x2 : ffffffc033fd0b58 
[  278.866447] x1 : 0000000000000000 x0 : 0000001400000014 
[  278.866479] 
[  278.866479] PC: 0xffffffc00075526c:
[  278.866499] 5268  9121e28136000aa0 913d840090005f60 941a291691214021 9128800090005f80 1400000d52800041 540001a17101927f 36000900b94e62c0 90005f609121e281
[  278.866584] 52a8  91214021913d8400 90005f80941a2909 2a1303e191298c00 1400003e941a2905 d2800021910003e0 f9400b809272c41c d5033bbff9000001 941a4b34d2802580
[  278.866662] 52e8  f900001ff9400b80 11000673d5033bbf f9425e6017ffffc9 d2804018d2802017 f2fbd5b79112e276 f8580414f2fbd5b8 91020001d1020294 540004c0eb16003f
[  278.866742] 5328  eb0102bff9400001 a948040254000121 f9000022f9000441 b9463261a9086017 b906326151000421 f940428197eeb925 d1020034aa1403e0 f9425e6017ffffef
[  278.866822] 5368  f85804019112e262 91020003d1020021 54000200eb02007f eb0302bff9400003 f940380054000121 5280002336080160 2a0303e2aa1503e1 97ffe7d1aa1303e0
[  278.866902] 53a8  f940402314000005 d1020061aa0103e0 5280000017ffffef a9425bf5a94153f3 a9446bf9a94363f7 a8c67bfda94573fb a9bb7bfdd65f03c0 a90153f3910003fd
[  278.866981] 53e8  a9025bf5aa0103f4 b00094232a0303f3 f94000019104a060 d2800001f90027a1 97fff9d9aa1403e0 9100e3a0aa0003f5 a9007c1faa0303f6 b9400040b40000a2
[  278.867060] 5428  b8402040b9003ba0 7100067fb803a3a0 3400013354000280 5400032171000a7f aa0203e352800060 5280000252800021 1400000e3900fba0 d2800003aa1403e0
[  278.867140] 5468  5280002152800002 2a0003f397ff6a47 528000c037f806a0 140000183900fba0 2a1303e1aa0203e3 aa1403e02a1303e2 2a0003f397ff6a3d 1400002a36f80220
[  278.867223] 
[  278.867223] LR: 0xffffffc0007550e4:
[  278.867242] 50e0  9121e021f0003a81 90005f609120e021 941a2978913d8400 9126400090005f80 361001401400000a 9121e021f0003a81 d0005f609120e021 941a296e911e2c00
[  278.867323] 5120  9126ac0090005f80 2a1303e0941a296b a9425bf5a94153f3 a8c47bfda94363f7 a9ba7bfdd65f03c0 a9025bf5910003fd a90153f3d0008a16 a90363f7aa0203f5
[  278.867402] 5160  b94e62c0aa0103f7 a90573fba9046bf9 f0003a8136100140 912140219121e021 911e2c00d0005f60 d0005fa0941a2953 941a29509114cc00 f9413eb9b4001135
[  278.867482] 51a0  aa1503e0b40010f9 aa0003f397fffa72 f9400000b4001060 eb0002bff9401000 f0003a9454000fe0 9121435a9121e29a aa1a03e1aa1703e2 9127340090005f80
[  278.867563] 51e0  f1001aff941a293d f10026ff540008a0 f1000aff54000ba0 b000865754000e21 90005f7aaa1a03f5 52800013913d875a f940a2f8528000bb f9403b2091096318
[  278.867643] 5220  1adb0e6036000ce0 6b00027f0b000800 b94e62c054000141 aa1503e136000100 941a2924aa1a03e0 91279c0090005f80 941a29202a1303e1 cb180000f940a2e0
[  278.867722] 5260  b94e62c0b7f80180 9121e28136000aa0 913d840090005f60 941a291691214021 9128800090005f80 1400000d52800041 540001a17101927f 36000900b94e62c0
[  278.867800] 52a0  90005f609121e281 91214021913d8400 90005f80941a2909 2a1303e191298c00 1400003e941a2905 d2800021910003e0 f9400b809272c41c d5033bbff9000001
[  278.867880] 52e0  941a4b34d2802580 f900001ff9400b80 11000673d5033bbf f9425e6017ffffc9 d2804018d2802017 f2fbd5b79112e276 f8580414f2fbd5b8 91020001d1020294
[  278.867962] 
[  278.867962] SP: 0xffffffc00eb8b830:
[  278.867981] b830  ffffffc0c1178a80 0000000000000000 0000000000000000 382e38373220205b 77205d3931363536 3230386766635f6c 656474656e5f3131 696669746f6e5f76
[  278.868060] b870  3a6c6c61635f7265 6e696c646e616820 2065746174732067 0000000000000000 0000000000000000 0000000000000000 ffffffc0819fae77 ffffffc033fd06a0
[  278.868140] b8b0  ffffffc000ea8000 ffffffc024014000 ffffffc001897000 0000000000000009 0000000000000001 ffffffc091a2a800 ffffffc000ea90c8 ffffffc016278878
[  278.868219] b8f0  0000000000000001 ffffffc00eb8b930 ffffffc0007551e4 ffffffc00eb8b930 ffffffc00075536c 0000000060000105 ffffffc0455142b8 0000000000000006
[  278.868296] b930  ffffffc00eb8b990 ffffffc000248770 00000000ffffffeb 0000000000000000 0000000000000009 ffffffc024014000 ffffffc01924dcc0 0000000000000001
[  278.868373] b970  ffffffbffc0bd8b0 ffffffc001980190 ffffffc016278878 0000000000000001 ffffffc00eb8b9d0 ffffffc00024883c 0000000000000009 ffffffc024014000
[  278.868453] b9b0  ffffffc0019da000 ffffffc00eb8ba58 ffffffc00142f11c ffffffbc02803040 ffffffc00eb8b9e0 ffffffc000be256c ffffffc00eb8ba00 ffffffc000be2618
[  278.868533] b9f0  ffffffc024014000 ffffffc00eb8bad8 ffffffc00eb8ba20 ffffffc000be276c ffffffc00eb8ba68 ffffffc00eb8bad8 ffffffc00eb8ba70 ffffffc000be42f4
[  278.868612] 
[  278.868632] Process khubd (pid: 76, stack limit = 0xffffffc00eb88058)
[  278.868649] Call trace:
[  278.868685] [<ffffffc00075536c>] wl_cfg80211_netdev_notifier_call+0x228/0x298
[  278.868722] [<ffffffc00024876c>] notifier_call_chain+0x4c/0x84
[  278.868749] [<ffffffc000248838>] raw_notifier_call_chain+0x10/0x1c
[  278.868780] [<ffffffc000be2568>] call_netdevice_notifiers+0x4c/0x5c
[... 9 lines cut ...]
[  278.869751] [<ffffffc0007a977c>] usb_unbind_interface+0x70/0x164
[  278.869785] [<ffffffc000612d98>] __device_release_driver+0x6c/0xbc
[  278.869809] [<ffffffc000612e08>] device_release_driver+0x20/0x38
[  278.869833] [<ffffffc0006126ec>] bus_remove_device+0x104/0x120
[  278.869856] [<ffffffc00060fddc>] device_del+0x110/0x174
[  278.869879] [<ffffffc0007a7540>] usb_disable_device+0xc4/0x234
[  278.869908] [<ffffffc00079f074>] usb_disconnect+0xb4/0x1ac
[  278.869933] [<ffffffc0007a0d60>] hub_thread+0x7f0/0x1138
[  278.869958] [<ffffffc000243488>] kthread+0xb8/0xcc
[  278.869986] Code: d1020034 17ffffef f9425e60 9112e262 (f8580401) 
[  278.870414] ---[ end trace 66d77106990ca416 ]---
[  278.886693] tusb320_intb_work: start
[  278.887228] Kernel panic - not syncing: Fatal exception
[  278.887261] CPU2: stopping

.....................................................................................................................

[  280.124379] scm_call failed: func id 0x82000301, arginfo: 0x1, args: 0x1, 0xffffffc000a77d10, 0xffffffc00eb8b2f0, 0xffffffc000a90d18, ret: -5, syscall returns: 0x0, 0x0, 0x0
[  280.124410] Rebooting in 5 seconds..
[  285.125423] Going down for restart now
[  285.125477] Warm reset for panic
[  285.126106] qcom,qpnp-power-on qpnp-power-on-12: PMIC@SID2: configuring PON for reset
[  285.126855] Causing a watchdog bite!

I have little experience in the development of kernel or kernel modules, so could you tell me in what direction should I look for the source of the problem? This is a feature of the Linux kernel, the Linux kernel for Android devices, some kind of conflict between the module configs and kernel configs or something else?

kimocoder, 2018-05-03:
The reboot error is not driver related. I also got this on other adapters (ath9k_htc).
So this has nothing to do with the driver, rather the power function on external devices / Android itself 

However, great you got it working on Android at all cause most adapters supported are high powered ones, not really great to be used om Android (NetHunter) in the first place due to demand of external USB hub and OTG-Y power.

alammer, 2018-05-03:
Thanks u for help!

Power in general is a problematic thing for me;) Work with Nexus 6P and cant find any OTG-Y power for his Type C port. However a fully charged device runs about an hour with asus usb-ac56 and running airodump-ng. I know that since the Oreo 6P get support nexmon for the internal adapter, but there is not enough time yet to test it.

kimocoder, 2018-05-03:
There are only 2 devices (phones) which got the kernel OTG-Y patch, Nexus 5x (bullhead) and another one which I can't remember right now, but not the Nexus 6P (angler) anyways.

alammer, 2018-05-04:
I will take note. You're doing a great job. Thanks again!

closed 2018-05-04 by alammer