aircrack-ng/rtl8812au #123: Help with Installing   AWUS036ACH Drivers

wynscott09, 2018-05-05:
Trying to run ./dkms-install.sh and I am receiving errors :

root@kali:~/rtl8812au# ./dkms-install.sh
About to run dkms install steps...
Error! DKMS tree already contains: rtl8812au-5.1.5
You cannot add the same module/version combo more than once.
Error! echo
Your kernel headers for kernel 4.13.0-kali1-amd64 cannot be found at
/lib/modules/4.13.0-kali1-amd64/build or /lib/modules/4.13.0-kali1-amd64/source.
Error! echo
Your kernel headers for kernel 4.13.0-kali1-amd64 cannot be found at
/lib/modules/4.13.0-kali1-amd64/build or /lib/modules/4.13.0-kali1-amd64/source.
Finished running dkms install steps.

I would appreciate your help.
Thank you.

kimocoder, 2018-05-05:
You need to install kernel headers, as described in the README.

apt-get install linux-headers-`uname -r`

wynscott09, 2018-05-05:
tried to run apt-get install linux-headers-`uname -r` 

root@kali:~/rtl8812au# apt-get install linux-headers-`uname -r`
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.13.0-kali1-amd64
E: Couldn't find any package by glob 'linux-headers-4.13.0-kali1-amd64'
E: Couldn't find any package by regex 'linux-headers-4.13.0-kali1-amd64'
:(

wynscott09, 2018-05-05:
I already installed the command below  :
sudo apt-get install build-essential
sudo apt-get install bc

but apt-get install linux-headers-uname -r is not working to me.

kimocoder, 2018-05-05:
Find package in apt with "apt-cache search linux-headers" or check that your apt actually for any sources with "cat /etc/apt/sources.list" at all

wynscott09, 2018-05-05:
"cat /etc/apt/sources.list" 

[image]

wynscott09, 2018-05-05:
apt-cache search linux-headers

[image]

kimocoder, 2018-05-05:
Well, can't find the v4.13 [here](https://http.kali.org/kali/pool/main/l/linux/) you should maybe consider upgrading to v4.14 or even v4.15.

wynscott09, 2018-05-05:
May i know the correct command/procedure for upgrading to v4.15.
really appreciate your help.
Thank you.

kimocoder, 2018-05-05:
apt-get install linux-image-4.15.0-kali3-amd64

Then reboot into it and follow the README with the "uname -r"  command

wynscott09, 2018-05-05:
linux-image-4.15.0-kali3-amd64 is already the newest version (4.15.17-1kali1).

kcdtv, 2018-05-05:
You can also use `(sudo) apt update && (sudo) apt full-upgrade " to update kali linux and it kernel. 
Now that you have the latest kernel installed repeat the installation process, the module will be placed in the correct folder

kimocoder, 2018-05-05:
You forget to reboot into kernel v4.15. reboot into it and install the headers package.

closed 2018-06-22 by kimocoder