aircrack-ng/rtl8812au #124: Can I use these instructions to install 8812au on the raspberry pi 3

danieleperera, 2018-05-09:
Hi guys i'm a newbie... and i'm trying to install 8812au on a raspberry pi moreover i have been trying for last 2 weeks and i found out that you have to enable 
$ sed -i -e 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile 
$ sed -i -e 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile

but in the makefile I can't find the CONFIG_PLATFORM_ARM_RPI so is there another way to install this driver on raspberry pi ? thank you

aircrack-ng, 2018-05-09:
Are you sure you are looking at the right project? The makefile contains both of those.

kimocoder, 2018-05-09:
Haven't tested it for RPI yet, it should work though. My plan is to test the RPI plattform anytime soon, so we'll leave the issue here for an update shortly.

danieleperera, 2018-05-10:
https://github.com/aircrack-ng/rtl8812au/blob/v5.2.9/Makefile 
This version hasn't got CONFIG_PLATFORM_ARM_RPI in Makefile

but this version has it 
https://github.com/aircrack-ng/rtl8812au/blob/v5.1.5/Makefile

so I cloned it with 

$ git clone https://github.com/aircrack-ng/rtl8812au/ --branch v5.1.5

sudo apt install raspberrypi-kernel-headers
sudo apt-get install build-essential
sudo apt-get install bc

and now its working even with monitor mode and packet injection mode thanks guys !!

kimocoder, 2018-05-10:
You should always use the v5.1.5 branch because of the support/patched added to it.
Great you got it working, haven't tested myself after added the RPI support not long ago.

closed 2018-05-11 by kimocoder

danieleperera, 2018-05-12:
I was wondering if buying a alfa APA-M25 could help to get a better signal  and if the code can enable 8 db for 2.4 ghz and 10 db for 5 ghz  by default o if i should run some other commands ?