aircrack-ng/rtl8812au #125: aarch64 isn't supported

charliebabbitt1988, 2018-05-10:
Can support for Odroid devices be added?

Trying to run "make" results in this error.

make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/3.14.79/build M=/root/rtl8812au  modules
make[1]: Entering directory '/usr/src/kernel'
Makefile:590: /usr/src/kernel/arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target '/usr/src/kernel/arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/kernel'
Makefile:1264: recipe for target 'modules' failed
make: *** [modules] Error 2

kimocoder, 2018-05-10:
It is added to Makefile, but it seems it should be be corrected somewhere.
will have a look at it om a couple lf days

kimocoder, 2018-05-10:
@charliebabbitt1988 which branch did you clone? "master", "v4.3.21", "v5.1.5" or "v5.2.9" ?

charliebabbitt1988, 2018-05-11:
I have tried both the "v5.1.5" and "v5.2.9" branches.  Both had a very similar error.  I can post output from "make" from the 2 branches if desired.

kimocoder, 2018-05-11:
Yes please

charliebabbitt1988, 2018-05-11:
v5.1.5
root@kali-arm64:~/v5.1.5/rtl8812au# make
make -j 4 ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/3.14.79/build M=/root/v5.1.5/rtl8812au  modules
make[1]: Entering directory '/usr/src/kernel'
Makefile:590: /usr/src/kernel/arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target '/usr/src/kernel/arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/kernel'
Makefile:1911: recipe for target 'modules' failed
make: *** [modules] Error 2

v5.2.9
root@kali-arm64:~/v5.2.9/rtl8812au# make
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/3.14.79/build M=/root/v5.2.9/rtl8812au  modules
make[1]: Entering directory '/usr/src/kernel'
Makefile:590: /usr/src/kernel/arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target '/usr/src/kernel/arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/kernel'
Makefile:1799: recipe for target 'modules' failed
make: *** [modules] Error 2

charliebabbitt1988, 2018-05-11:
Looks like I might be missing the linux-headers for the Odroid C2.  This looks like a lead worth investigating.  https://archlinuxarm.org/packages/aarch64/linux-odroid-c2-headers

"uname -r" shows 3.14.79 for Kali 2018.2 which is a match with what can be found at that link.

Let me see what I can make happen.

kimocoder, 2018-05-11:
Yes, this looks like your headers package is missing. I'm not familiar to Odroid, but try finding the headers package for that kernel and install it and try again.

charliebabbitt1988, 2018-05-13:
Actually, I found this.

https://github.com/offensive-security/kali-arm-build-scripts/issues/117

This is the same issue that I am having.

If you look at the bottom post there, it says that a one line change to "SUBARCH" can solve this problem.

A simple change to the Makefile is probably enough to solve this problem.

Let me know if I should submit a pull request.

kimocoder, 2018-05-13:
Aaaah nice catch. Haven't seen that one until now. Could you test ut, see if that change actually works, then report back?

kimocoder, 2018-05-13:
The changes have been made, as seen here https://github.com/aircrack-ng/rtl8812au/commit/d628a7441c6066a2e302104b7f4f99bcd81a111f

Use "git pull" to fetch the updated and try building it again, since you actually got the hardware to test with.
Since "strev" was the one that posted it in the link above I suppose it's a working patch.

Let med know if that works and report back.

charliebabbitt1988, 2018-05-13:
The change to the Make file setting of "SUBARCH" on line 1071 didn't work.  This was the result.

make -j 4 ARCH=arm64 CROSS_COMPILE= -C /lib/modules/3.14.79/build M=/root/v5.1.5                             /rtl8812au  modules
make[1]: Entering directory '/usr/src/kernel'

  WARNING: Symbol version dump /usr/src/kernel/Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_cmd.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_security.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_debug.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_io.o
scripts/Makefile.build:308: recipe for target '/root/v5.1.5/rtl8812au/core/rtw_i                             o.o' failed
scripts/Makefile.build:308: recipe for target '/root/v5.1.5/rtl8812au/core/rtw_c                             md.o' failed
scripts/Makefile.build:308: recipe for target '/root/v5.1.5/rtl8812au/core/rtw_s                             ecurity.o' failed
scripts/Makefile.build:308: recipe for target '/root/v5.1.5/rtl8812au/core/rtw_d                             ebug.o' failed
Makefile:1278: recipe for target '_module_/root/v5.1.5/rtl8812au' failed
make[1]: Leaving directory '/usr/src/kernel'
Makefile:1911: recipe for target 'modules' failed

kimocoder, 2018-05-14:
Like it says, the "Module.symvers" is missing from your kernel directory. You should have that fixed though.

charliebabbitt1988, 2018-05-14:
Prior to running "make", if I execute this first,

cp .config ../odroid-c2.config
make mrproper
cp ../odroid-c2.config .config
make prepare
make modules_prepare

and with the same edit to the Makefile, then it compiles fine.  Here is the output:

root@kali-arm64:~/v5.1.5/rtl8812au# make
make -j 4 ARCH=arm64 CROSS_COMPILE= -C /lib/modules/3.14.79/build M=/root/v5.1.5/rtl8812au  modules
make[1]: Entering directory '/usr/src/kernel'

  WARNING: Symbol version dump /usr/src/kernel/Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_cmd.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_security.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_debug.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_io.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_mlme.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_mi.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_vht.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_rf.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_recv.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_ap.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_xmit.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_p2p.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_tdls.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_br_ext.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_iol.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_sreset.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_btcoex.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_beamforming.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_odm.o
  CC [M]  /root/v5.1.5/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/osdep_service.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/os_intfs.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/usb_intf.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/usb_ops_linux.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/ioctl_linux.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/xmit_linux.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/mlme_linux.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/recv_linux.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/wifi_regd.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/rtw_android.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/rtw_radiotap.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/rtw_proc.o
  CC [M]  /root/v5.1.5/rtl8812au/os_dep/linux/ioctl_mp.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/hal_intf.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/hal_com.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/hal_com_phycfg.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/hal_phy.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/hal_dm.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/hal_btcoex.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/hal_mp.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/hal_mcc.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/hal_hci/hal_usb.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/led/hal_usb_led.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/HalPwrSeqCmd.o
[... 48 lines cut ...]
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/rtl8812a/halphyrf_8812a_ce.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/rtl8812a/phydm_regconfig8812a.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/rtl8812a/phydm_rtl8812a.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/txbf/haltxbfjaguar.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_fw.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_mac.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_bb.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_rf.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/rtl8821a/halphyrf_8821a_ce.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/rtl8821a/phydm_regconfig8821a.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/rtl8821a/phydm_rtl8821a.o
  CC [M]  /root/v5.1.5/rtl8812au/hal/phydm/rtl8821a/phydm_iqk_8821a_ce.o
  CC [M]  /root/v5.1.5/rtl8812au/platform/platform_ops.o
  CC [M]  /root/v5.1.5/rtl8812au/core/rtw_mp.o
  LD [M]  /root/v5.1.5/rtl8812au/8812au.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /root/v5.1.5/rtl8812au/8812au.mod.o
  LD [M]  /root/v5.1.5/rtl8812au/8812au.ko
make[1]: Leaving directory '/usr/src/kernel'
root@kali-arm64:~/v5.1.5/rtl8812au# make install
install -p -m 644 8812au.ko  /lib/modules/3.14.79/kernel/drivers/net/wireless/
/sbin/depmod -a 3.14.79

I will do some testing soon to see how/if it works.

kimocoder, 2018-05-14:
This looks more promising. Report back

charliebabbitt1988, 2018-05-16:
I tested the result with an AWUS036ACH.

Output from lsusb:
Bus 001 Device 011: ID 093a:2530 Pixart Imaging, Inc. 
Bus 001 Device 010: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
Bus 001 Device 009: ID 0c45:7603 Microdia 
Bus 001 Device 008: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Output from lsmod:
Module                  Size  Used by
8812au               1704209  0
spidev                  6459  0
joydev                 11047  0
cfg80211              476295  1 8812au
spi_gpio                6714  0
spi_bitbang             4071  1 spi_gpio
w1_gpio                 3785  0
wire                   25671  1 w1_gpio
meson_gpiomem           4024  0
gxbb_wdt                3427  0
ip_tables              17005  0
x_tables               22810  1 ip_tables
autofs4                28111  2

Output from dkms-install.sh:
DKMS make.log for rtl8812au-5.1.5 for kernel 3.14.79 (aarch64)
Tue May 15 22:40:36 CDT 2018
make -j 4 ARCH=arm64 CROSS_COMPILE= -C /lib/modules/3.14.79/build M=/var/lib/dkms/rtl8812au/5.1.5/build  modules
make[1]: Entering directory '/usr/src/kernel'

  WARNING: Symbol version dump /usr/src/kernel/Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/os_dep/linux/mlme_linux.o
[... 184 lines cut ...]
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/hal/phydm/phydm_kfree.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/hal/phydm/phydm_ccx.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/hal/phydm/rtl8814a/halhwimg8814a_bb.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/hal/phydm/rtl8814a/halhwimg8814a_mac.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/hal/phydm/rtl8814a/halhwimg8814a_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/hal/phydm/rtl8814a/halhwimg8814a_fw.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/hal/phydm/rtl8814a/phydm_iqk_8814a.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/hal/phydm/rtl8814a/phydm_regconfig8814a.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/hal/phydm/rtl8814a/halphyrf_8814a_ce.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/hal/phydm/rtl8814a/phydm_rtl8814a.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/hal/phydm/txbf/haltxbf8814a.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/platform/platform_ops.o
  CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_mp.o
/var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_mp.c: In function 'rtfloor':
/var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_mp.c:36:12: error: '-mgeneral-regs-only' is incompatible with floating-point argument
 static int rtfloor(float x) {
            ^~~~~~~
scripts/Makefile.build:308: recipe for target '/var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_mp.o' failed
make[2]: *** [/var/lib/dkms/rtl8812au/5.1.5/build/core/rtw_mp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
Makefile:1278: recipe for target '_module_/var/lib/dkms/rtl8812au/5.1.5/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl8812au/5.1.5/build] Error 2
make[1]: Leaving directory '/usr/src/kernel'
Makefile:1911: recipe for target 'modules' failed
make: *** [modules] Error 2

kimocoder, 2018-05-16:
So the 8812au builds, but not the 8814au? Will take a look at it right away

charliebabbitt1988, 2018-05-17:
I tried to execute "make RTL8814=1" and I got an identical result in the output.

I might switch distros and see if I can get a better result.  Ubuntu 16 from Hardkernel could be a good option.  As long as the build shows "uname -m" as arm64 instead of aarch64, then that could work.

charliebabbitt1988, 2018-05-17:
For clarification, the rtl8812au driver seems to function, but I can't get it into monitor mode.

It at least shows up as an interface in the OS as wlan0. 

I'll try this one out and see what happens.

https://github.com/commx/rtl8812au

igorpecovnik, 2018-06-02:
> I might switch distros and see if I can get a better result.

IMO it is not distro related. Building fails on arm64 with kernel: 3.14.y and 4.4.y while it compiles on 4.14.y for example.

@kimocoder You can also add to the main readme that this driver works on all versions of Armbian (well its already implemented), where kernel version is higher than 3.14, with already mentioned exceptions. They apply to Odroid C2, Rock64, Espressobin, Odroid N1 and all upcoming arm64 boards with (legacy) kernels, usually 4.4 or 4.9 which is possibly affected by this problem.

I could not resolve this but hope those findings will help.

kimocoder, 2018-06-05:
The issue is mentioned [here](https://forum.armbian.com/topic/6417-troubles-with-aarch64-compiler-or/) too.. looking into it

kimocoder, 2018-06-05:
There are some slight differences in different forks.. as this one https://github.com/zebulon2/rtl8814au/blob/master/core/rtw_mp.c#L36

You may try fiddle around with it? It would be nice to get it working

madscientist42, 2018-06-21:
It does _**seem**_ to build correctly (haven't tried _**YET**_ with HW) on my OE build based off of the meta-rtlwifi and meta-pha layers in my repo set here.  Having said this, I'm running against 4.16 on a NanoPI Neo2 or Neo-Plus2 in this configuration.

igorpecovnik, 2018-06-21:
> I'm running against 4.16

That works fine for me too, but older (4.4 or 4.9) doesn't work.

kimocoder, 2018-06-29:
Closing this for now, started working on v5.2.20. For 8812 users only atm, but adding 8814/8821 HAL is on the TODO list. 

v5.2.9 should not be used, Realtek has put in their changelog. Besides that, we've done none work on the v5.2.9, bit the v5.2.20 we've added frame injection. Needs more testers to it too ;)

closed 2018-06-29 by kimocoder

madscientist42, 2018-06-29:
Will look into adding the later version into a experimental branch of my OE
metadat layer.