aircrack-ng/rtl8812au #126: Can't change reg domain on rtl8812au chipset

alammer, 2018-05-11:
Hi!

Does https://github.com/aircrack-ng/rtl8812au/issues/108 and https://github.com/aircrack-ng/rtl8812au/commit/ec103e0fa66aaac4d7349bffbfd73e4f3256452e means that all the limitations for the driver in v.5.1.5 branch are currently caused solely by the settings of the kernel but not of the module itself?

kimocoder, 2018-05-11:
It was hardcoded in the driver, but has been removed as it should not be hardcoded at all.
anyway, the txpower doesn't change because of the power issue you may find in another report.

alammer, 2018-05-11:
I don't say about power problems. Now iw reg get return:

iw reg get
country 00:
	(2402 - 2472 @ 40), (3, 20)
	(2457 - 2482 @ 40), (3, 20), PASSIVE-SCAN, NO-IBSS
	(2474 - 2494 @ 20), (3, 20), NO-OFDM, PASSIVE-SCAN, NO-IBSS
	(5170 - 5250 @ 40), (3, 20), PASSIVE-SCAN, NO-IBSS
	(5735 - 5835 @ 40), (3, 20), PASSIVE-SCAN, NO-IBSS

iw reg set X return error with code (-7).

Does I have to only patching my kernel according to https://github.com/aircrack-ng/rtl8812au/issues/108/ for set reg domain? Or should I do something else with the module settings or sources?

kimocoder, 2018-05-11:
It should do as other driver should, but the error code does look like something is wrong.

I will take a look at it again this weekend, what happens over here. Maybe it needs some modifications.

Some people use "iw reg set BO" (for Bolivia), which enables the driver to run on 30dBm but there is a issue to the power / power limitation at the moment which restricts the power (a bug).

kimocoder, 2018-05-11:
Over here the output is:

country NO: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
	(5470 - 5795 @ 160), (N/A, 26), (0 ms), DFS
	(5815 - 5850 @ 35), (N/A, 33), (0 ms), DFS
	(17100 - 17300 @ 200), (N/A, 20), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

which also are correct for Norway..

alammer, 2018-05-15:
THX! The problem was completely on the kernel side. After some steps from https://github.com/aircrack-ng/rtl8812au/issues/108 all channels from 8812au sources available for adapter by default and work fine.

closed 2018-05-15 by alammer