aircrack-ng/rtl8812au #127: What's the best way to apply these settings at boot?

francoism90, 2018-05-12:
e.g. enable monitoring + txpower.

Is it simple by creating a script, don't autostart `NetworkManager.service` at boot?

Thanks.

FallenArchangel, 2018-05-17:
I just created a file called monitor.sh and placed it in my home directory.
Inside it I chained all the commands together: 

sudo ip link set wlan0 down && sudo iw dev wlan0 set type monitor && sudo ip link set wlan0 up && sudo iw wlan0 set txpower fixed 3000

Save the file, then do:
Chmod +X monitor.sh

And then any time I turn on my laptop I can just run:
./monitor.sh
And it will do everything for me.  
I'm sure there is a much better way of doing this, but until someone else replies with a better way, maybe this will help. Just remember to change wlan0 to wlan1 or wlan2 or whatever you are using.

francoism90, 2018-09-25:
```
[Unit]
Description=Set TX-Power and Monitor-mode %I
After=systemd-udevd.service network-pre.target systemd-sysctl.service
Before=network.target multi-user.target shutdown.target
Wants=network.target

[Service]
Type=oneshot
RemainAfterExit=yes
#ExecStart=sh -c 'ip link set %i down; iw dev %i set type monitor; ip link set %i up; iw %i set txpower fixed 3000'
ExecStart=sh -c 'iw %i set txpower fixed 3000'

[Install]
WantedBy=multi-user.target
```

Enable by `# systemctl enable 88XXau@interface.service'