aircrack-ng/rtl8812au #128: Unable to install on Nethunter. (S7 Exynos)

FallenArchangel, 2018-05-17:
I'm not sure if this is an issue with Nethunter or these drivers, so I'm posting just in case.

Whenever I try to install the realtek-rtl88xxau-dkms package I get the following output:
[installdkms.txt](https://github.com/aircrack-ng/rtl8812au/files/2013582/installdkms.txt)

And when I try to run make RTL8814=1, I get this:
[make.txt](https://github.com/aircrack-ng/rtl8812au/files/2013586/make.txt)

I was able to get this working on a raspberry pi zero by installing the linux-headers and adding ARCH=arm, but I don't seem to be able to find the headers I need for this(linux-headers-3.18.44-nethunter-herolte-xx-1.5), so that's not an option.

Is there anything I can do to get this working?  Thanks.

alammer, 2018-05-18:
All that I could with a bunch of rtl8812(14)au.ko and Nethunter on my Nexus 6P is to find the kernel sources as close as possible (thanks to https://github.com/jcadduono) to the kernel of Nethunter's working version, modify their build configs to a stable running state, build the kernel image  itself, make  kernel headers and modules prepare, after all build module for 8812/8814. Then build custom Kali Nethunter with my custom kernel image and nethunter-installer and copy 8812 module on phone manually. I hope you find an easier way:)

kimocoder, 2018-05-18:
@FallenArchangel you are missing the kernel headers. you need to find them prior to your device.

closed 2018-12-23 by kimocoder

BeingDatta, 2020-04-09:
> I'm not sure if this is an issue with Nethunter or these drivers, so I'm posting just in case.
>
> Whenever I try to install the realtek-rtl88xxau-dkms package I get the following output: [...]

Hey there if your kernel headers are not available in "apt-cache search uname-r" you can built it by yourself using your android kernel source code and cross compiler just follow the process on this link----"https://github.com/aircrack-ng/rtl8188eus/issues/31#issue-590043142"