aircrack-ng/rtl8812au #129: support WEP

ZeroChaos-, 2018-05-21:
WEP is unsupported in client mode

kimocoder, 2018-06-22:
@ZeroChaos- for the 8812 or the 8814 adapter?

ZeroChaos-, 2018-06-23:
Both? It presently fails on both in my testing.

kimocoder, 2018-06-23:
Checking ut tomorrow, haven't really used WEP for some years

kimocoder, 2018-06-24:
@ZeroChaos- WEP (in client mode) works just fine here? At least using 8814au chipset.
May try the 8812au when it arrives too, since their code (in the same driver) are slighty different

kcdtv, 2018-06-24:
As expected it worked for me too with chip 8812au:  I am connected right now through a WEP 64 network that i will shut down immediately. :smile_cat:  
[image]
Dropping WEP support in client mode wouldn't be anyway such a lost... ...But it works.

kimocoder, 2018-06-24:
I was using the v5.1.5 for testing btw. Branches below is not maintained and contains too much problems that they should not be used. 

Moving it all over to the v5.2.20 branch soon which fixes lots more issues.

kimocoder, 2018-07-24:
WEP tested & working on both 8812, 8814 and 8821 ... open another issue if the problem occurs again, but can't seem to find an issue about it?

closed 2018-07-24 by kimocoder