aircrack-ng/rtl8812au #130: we need working hostapd.conf

Karlodun, 2018-05-21:
It took me around 8h last time to have a well working hostapd.conf.
Then my SSD screwed up and I had to restart this weekend. After spending few hours I have a non-working one (for whatever reason) conf file  or example in README.
some of the features seem not to work properly, or I am misconfiguring them.....
iw config shows, [TX-STBC] is there, but hostapd says the driver does not support it....
Hopefully I'll be able to find a good and working example.

` # NOT WORKING FOR UNKNOWN REASON (yet)
interface=wlan0
#bridge=br0
driver=nl80211

logger_syslog=-1
logger_syslog_level=2
logger_stdout=-1
logger_stdout_level=2
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

##### IEEE 802.11 related configuration #######################################
ssid=mkg5
hw_mode=a
channel=40
auth_algs=3

country_code=DE
ieee80211d=1
ieee80211h=1

#local_pwr_constraint=3

beacon_int=100
dtim_period=2

max_num_sta=255
rts_threshold=-1
fragm_threshold=-1

# 0 = accept unless in deny list # 1 = deny unless in accept list
# 2 = use external RADIUS server (accept/deny lists are searched first)
macaddr_acl=0
#accept_mac_file=/etc/hostapd.accept
#deny_mac_file=/etc/hostapd.deny

ignore_broadcast_ssid=0

# can this one be moved down?
wmm_enabled=1
wmm_ac_bk_cwmin=4
wmm_ac_bk_cwmax=10
wmm_ac_bk_aifs=7
wmm_ac_bk_txop_limit=0
wmm_ac_bk_acm=0
wmm_ac_be_aifs=3
wmm_ac_be_cwmin=4
wmm_ac_be_cwmax=10
wmm_ac_be_txop_limit=0
wmm_ac_be_acm=0

wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=3
wmm_ac_vi_cwmax=4
wmm_ac_vi_txop_limit=94
wmm_ac_vi_acm=0

wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0

##### IEEE 802.11n related configuration ######################################
ieee80211n=1
ht_capab=[HT40-][GF][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40][MAX-ASMDU-7935]

##### IEEE 802.11ac related configuration #####################################
ieee80211ac=1
# Tx STBC: [TX-STBC-2BY1]
vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][SU-BEAMFORMEE][HTC-VHT]
vht_oper_chwidth=1

##### WPA/IEEE 802.11i configuration ##########################################
wpa=2
wpa_key_mgmt=WPA-PSK WPA-EAP
wpa_pairwise=CCMP
wpa_passphrase=<passphraze>
`

kimocoder, 2018-05-21:
If you give me some more time, I'll switch to a newer I/O control mechanisme instead of the outdated wireless extensions. This will be needed to run standard hostapd. Work in progress.

kcdtv, 2018-05-24:
i tried briefly with [create_ap](https://github.com/oblique/create_apl) and it worked fine in **bgn** (with a warning due toi the fact that it does "not fully support virtual interfaces")  but it failed when I tried to enable **ac** mode
```
create_ap wlan1 eth0 wifi 12345678 --ieee80211ac
WARN: Your adapter does not fully support AP virtual interface, enabling --no-virt
Config dir: /tmp/create_ap.wlan1.conf.E1LTV4Cs
PID: 13457
Sharing Internet using method: nat
hostapd command-line interface: hostapd_cli -p /tmp/create_ap.wlan1.conf.E1LTV4Cs/hostapd_ctrl
Configuration file: /tmp/create_ap.wlan1.conf.E1LTV4Cs/hostapd.conf
Using interface wlan1 with hwaddr 00:c0:ca:xxxxxxxapparently  and ssid "wifi"
Failed to set beacon parameters
Interface initialization failed
wlan1: interface state UNINITIALIZED->DISABLED
wlan1: AP-DISABLED 
wlan1: Unable to setup interface.
wlan1: interface state DISABLED->DISABLED
wlan1: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan1 wasn't started
nl80211: deinit ifname=wlan1 disabled_11b_rates=0

Error: Failed to run hostapd, maybe a program is interfering.

Doing cleanup.. done
```
Apparently it "failed to set beacon parameter"
When it worked, in b/g/n mode, beacons where set like this (100):
```
beacon_int=100
ssid=wifi
interface=wlan1
driver=nl80211
channel=1
ctrl_interface=/tmp/create_ap.wlan1.conf.lIYZPwJc/hostapd_ctrl
ctrl_interface_group=0
ignore_broadcast_ssid=0
ap_isolate=0
hw_mode=g
wpa=3
wpa_passphrase=12345678
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP CCMP
rsn_pairwise=CCMP
```
**creaste_ap** is a good program that usually works fine so maybe it can help to debugg this issue using it as a starting point. By the way, i used an awus036ach with **rtl8812au** and driver were the ones from  branch **5.1.5.**  cheers

kimocoder, 2018-05-24:
Yes, I had some few hours looking into the code and the driver runs on cfg80211/nl80211 which supports hostapd (or should), but the virtual Interface issue really annoyes me, the code is there but it clearly got some issue.

Karlodun, 2018-05-26:
I'm going to (re)start my experiments today.
Following info so far: Adapter does not support virtual interface, adapter does not support mode mixes.
THUS: EITHER 2.4Ghz a/b/g/n or 5 ac/n.

Karlodun, 2018-05-26:
This simple version works (but is not optimized for our card, nor optimized QoS):
interface=wlan0
hw_mode=a
channel=36
ieee80211d=1
country_code=DE
ieee80211n=1
ieee80211ac=1
wmm_enabled=1
ssid=mkg_ac
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
wpa_passphrase=jupiter1221

ggerla, 2018-06-01:
Are you sure that this configuration works?
How many hours (or days) do you test this configuration?
With a similar configuration (same channel, country code IT) but with VHT enabled, I had frequent disconnection... I'm using AWUS1900 for 5ghz AP and AWUS036AC for 2.4Ghz AP. Both have 5.1.5 version of driver. The 2.4Ghz seems more stable.
The prblem is that, after disconnection, AP and SSID are still active but client is not able to connect because there is a timeout on handshaking

IEEE 802.1X: unauthorizing port
WPA: sending 1/4 msg of 4-Way Handshake
....
WPA: EAPOL-Key timeout

I'm trying to do a lot of test changing parameter in the hostapd.conf but until now without success.

Karlodun, 2018-06-01:
The hostapd activates with this config. It is not optimal and is not stable. The stability issue is probably an issue of driver. In my case the connection drops very often, and ssid disappears for seconds. A reconnection is not always possible. In addition the signal is lower than that of my tenda ac9.
This and next two weeks I will not have enough time to test everything, still need to setup DHCP and DNS server for full AP mode.

ggerla, 2018-06-01:
so we are experiencing the same problem...
Following you can find my hostapd configuration that improve performance and is stable for some hours.

interface=wlx00c0ca954d8f
driver=nl80211
country_code=IT
ignore_broadcast_ssid=0
logger_syslog=-1
logger_syslog_level=1
bridge=wlan

ssid=YYYYY

wpa=2
wpa_passphrase=XXXXXX
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

hw_mode=a
channel=36
ieee80211n=1
ieee80211ac=1

ht_capab=[LDPC][HT40+][SHORT-GI-20][SHORT-GI-40]
vht_capab=[RXLDPC][SHORT-GI-80][RX-STBC-1][MAX-A-MPDU-LEN-EXP7]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

kimocoder, 2018-08-06:
This should be a working one.. not tested but..

ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
interface=wlan2
driver=nl80211
bridge=br0

### IEEE 802.11
ssid=<SOMESSID>
hw_mode=a
channel=36
max_num_sta=128
auth_algs=1

### DFS
country_code=DE
ieee80211d=1
ieee80211h=1

### IEEE 802.11n
ieee80211n=1
ht_capab=[HT20][HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]

### IEEE 802.11ac
ieee80211ac=1
#vht_oper_chwidth=1
#vht_capab=[HT40-]
#vht_oper_centr_freq_seg0_idx=46

### IEEE 802.11i
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_passphrase=<PASSPHRASE>
rsn_pairwise=CCMP

### WMM
wmm_enabled=1
uapsd_advertisement_enabled=1
wmm_ac_bk_cwmin=4
wmm_ac_bk_cwmax=10
wmm_ac_bk_aifs=7
wmm_ac_bk_txop_limit=0
wmm_ac_bk_acm=0
wmm_ac_be_aifs=3
wmm_ac_be_cwmin=4
wmm_ac_be_cwmax=10
wmm_ac_be_txop_limit=0
wmm_ac_be_acm=0
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=3
wmm_ac_vi_cwmax=4
wmm_ac_vi_txop_limit=94
wmm_ac_vi_acm=0
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0

### TX queue parameters
tx_queue_data3_aifs=7
tx_queue_data3_cwmin=15
tx_queue_data3_cwmax=1023
tx_queue_data3_burst=0
tx_queue_data2_aifs=3
tx_queue_data2_cwmin=15
tx_queue_data2_cwmax=63
tx_queue_data2_burst=0
tx_queue_data1_aifs=1
tx_queue_data1_cwmin=7
tx_queue_data1_cwmax=15
tx_queue_data1_burst=3.0
tx_queue_data0_aifs=1
tx_queue_data0_cwmin=3
tx_queue_data0_cwmax=7
tx_queue_data0_burst=1.5

kimocoder, 2018-08-06:
Another note.. use the v5.2.20 branch which runs the v5.2.20.2 driver, instead of the v5.1.5.
It got improved stability. Reporting back here how it works would be great. Thanks

Karlodun, 2018-08-09:
I'll get back to this when enough time to restart my OpenSUSE based firewalled switch :-)
will be a month or two

ThijsWithaar, 2018-09-05:
The following works for me. It's the most minimal config I could find to enable 80MHz bandwidth.
Gives me up to 12MByte/sec, probably limited by the Raspberry pi.

```
interface=wlx6c198f995532
driver=nl80211
ssid=YourSsid

# 5 Ghz
hw_mode=a
ieee80211n=1
ieee80211ac=1
# This enables radar detection and DFS support
ieee80211h=1
# This advertises the country_code
ieee80211d=1
country_code=US

channel=100
vht_oper_centr_freq_seg0_idx=106
vht_oper_chwidth=1
ht_capab=[HT40+]

wpa=2
wpa_passphrase=YourSecret
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
```

Karlodun, 2018-10-06:
well, I had several "minimal" setups... I was seeking for the best optimal one.
Few weeks ago I purchased a new'n cheap Laptop for Linux stuff, and yesterday the new ssd drive arrived. Hope I'll restart with my linux stuff soon. (sorry for off-topic)

kimocoder, 2019-02-15:
Documentation and schematics has been added to the folder with the same name, "documentation" also I put some Realtek examples there too. :+1:

closed 2019-02-15 by kimocoder

Medullitus, 2022-02-04:
> Documentation and schematics has been added to the folder with the same name, "documentation" also I put some Realtek examples there too. 👍

You talk too much but don't anything to bring vif to rtl8812au!!!