aircrack-ng/rtl8812au #132: package loss + non working monitoring channel

atar-axis, 2018-06-01:
hey there,
first of all: thank you for your hard work to everyone!
unfortunately i have two problems:

I am loosing a lot of packages in monitoring mode. the reason seems to be that the adapter is hopping from one channel to the next one. that's normal I guess, but the problem is that I really need ALL packages on exactly ONE channel. therefore, I tried to set the channel using
`airmon-ng start <interface> <channel>`.

Well, that doesnt work as you may already know (according to the README). I therefore tried to set the channel by `iw wlp0s20u1 set channel <channel#>` which does nor result in an error, but does not work either... airodump does "stand still" for around a minute or so and then the adapter starts "hopping" again...

I am really clueless...

kimocoder, 2018-06-22:
@atar-axis which chipset is this? 8812au or the 8814au? which adapter?

atar-axis, 2018-06-22:
I sent back the device since I am very busy at the moment, thank you anyway!!

closed 2018-07-24 by kimocoder