aircrack-ng/rtl8812au #134: 5Ghz networks disappearing after sleep/wake cycles

RJVB, 2018-06-17:
I'm seeing an issue with this driver where 5Ghz networks are no longer detected after a few sleep/wake cycles. They may come back if I unload/reload the driver often enough, only to disappear again after another sleep/wake cycle or two.

I wonder if this is related to the fact that reported network strengths are so much lower than with the older driver version from umdlife/rtl8812au_rtl8821au?

Plugging in my 8812 adapter makes this appear in the syslog:
```
Jun 17 10:42:25 bola kernel: [578468.805423] usb 1-2: new high-speed USB device number 18 using xhci_hcd
Jun 17 10:42:25 bola kernel: [578468.975954] usb 1-2: New USB device found, idVendor=0bda, idProduct=a811
Jun 17 10:42:25 bola kernel: [578468.975959] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 17 10:42:25 bola kernel: [578468.975961] usb 1-2: Product: 802.11ac WLAN Adapter 
Jun 17 10:42:25 bola kernel: [578468.975963] usb 1-2: Manufacturer: Realtek 
Jun 17 10:42:25 bola kernel: [578468.975965] usb 1-2: SerialNumber: 00e04c000001
Jun 17 10:42:26 bola mtp-probe: checking bus 1, device 18: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
Jun 17 10:42:26 bola mtp-probe: bus: 1, device: 18 was not an MTP device
Jun 17 10:42:26 bola kernel: [578469.196493] 8812au 1-2:1.0 wlan3: renamed from wlan0
Jun 17 10:42:26 bola NetworkManager[3916]: <info> rfkill33: found WiFi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/ieee80211/phy15/rfkill33) (driver 8812au)
Jun 17 10:42:26 bola NetworkManager[3916]:    SCPlugin-Ifupdown: devices added (path: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/net/wlan3, iface: wlan3)
Jun 17 10:42:26 bola NetworkManager[3916]:    SCPlugin-Ifupdown: device added (path: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/net/wlan3, iface: wlan3): no ifupdown configuration found.
Jun 17 10:42:26 bola NetworkManager[3916]: <info> (wlan3): using nl80211 for WiFi device control
Jun 17 10:42:26 bola NetworkManager[3916]: <info> (wlan3): driver supports Access Point (AP) mode
Jun 17 10:42:26 bola kernel: [578469.235868] systemd-udevd[19460]: renamed network interface wlan0 to wlan3
Jun 17 10:42:26 bola NetworkManager[3916]: <info> (wlan3): new 802.11 WiFi device (driver: '8812au' ifindex: 18)
Jun 17 10:42:26 bola NetworkManager[3916]: <info> (wlan3): exported as /org/freedesktop/NetworkManager/Devices/32
Jun 17 10:42:26 bola NetworkManager[3916]: <info> (wlan3): device state change: unmanaged -> unavailable (reason 'managed') [10 20 2]
Jun 17 10:42:26 bola NetworkManager[3916]: <info> (wlan3): bringing up device.
```

RJVB, 2018-06-17:
Signal strengths may indeed be at play here. When I filed this issue I couldn't find my 5Ghz network, provided by the same modem as the 2.4Ghz network I was on. That network was then reported to be at 32%. A couple of driver unload/reloads (including the older 4.3.20 driver) later I'm back on the 5Ghz network; it and the 2.4Ghz network are now reported at 56% and 50% respectively.

kimocoder, 2018-06-23:
It says 

"Jun 17 10:42:26 bola NetworkManager[3916]: <info> rfkill33: found WiFi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/ieee80211/phy15/rfkill33) (driver 8812au)"

which indicated that "rfkill" may not be playing along with you here. you may check if it's blocked with
the "rfkill list" command and unblock it with "rfkill unblock #number found here"

RJVB, 2018-06-23:
>which indicated that "rfkill" may not be playing along with you here

Why? Isn't that just a log message to indicate that rfkill found a new device (= when I insert the USB adapter)?

kimocoder, 2018-07-24:
@RJVB try the v5.2.20 branch, report back if that fixes the issue or not so this may be closed. Thanks

RJVB, 2018-07-24:
Sadly I still get the same problem. I have in fact all but stopped using the 8812au for now because I'm fed up trying to get it to reconnect.

kimocoder, 2019-02-15:
I have disabled some sleep/wake cycles. Try updating the driver and see if the problem persists :+1:

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)