aircrack-ng/rtl8812au #135: Unreasonably low reported network strengths

RJVB, 2018-06-17:
Reported network strengths are much lower with this driver than with the older driver version from umdlife/rtl8812au_rtl8821au. For instance, the internal Intel WiFi in my laptop reports 51% for the 2.4Ghz network across the room at maybe 5m, while the external 8812au USB adapter reports only 32% but I've seen differences of more than 2x lower.

Can there be some scale factor that's involved here? Somewhere someone suggested this might be for regulatory reasons (updated w.r.t. the older driver version) but I don't really see how those could affect the receiver function?

stephanstricker, 2018-06-22:
I have the same issue. I am running the driver in VirtualBox on an Imac host. The iMac host see's twice the access points so it is not an issue of USB power. I increased TX power to 30 but this does not change anything.

kimocoder, 2018-06-22:
This is related to https://github.com/aircrack-ng/rtl8812au/issues/77 so continue om that thread instead of both two.

closed 2018-06-22 by kimocoder