aircrack-ng/rtl8812au #154: Ubuntu 16.04 AWUS036ACH no packets captured with monitor mode

fightingyuman, 2018-08-03:
I'am use linux 16.04 with 4.15.0 linux kernel, with AWUS036ACH adapter, the linux kernel information is as below:
> Linux yuman-unbuntu16 4.15.0-29-generic #31~16.04.1-Ubuntu SMP Wed Jul 18 08:54:04 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

I use this in a virtual machine with VMware.

I use the following command to set the adapter to monitor mode and kill some process:
```
ifconfig wlx00c0ca96611b down
iw dev wlx00c0ca96611b set monitor none
ifconfig  wlx00c0ca96611b up
airmon-ng check kill
```

I can't capture any packets with tcpdump nor airodump-ng with that adapter.

When I use command **iwconfig**, I get the following information:
> wlx00c0ca96611b  IEEE 802.11  Mode:Monitor  Frequency:2.412 GHz  Tx-Power=18 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

The Frequency changes when airodump-ng is on.

Any help is appreciated.

kimocoder, 2018-08-06:
Some of your steps there are.. just wrong.

1. airmon-ng check kill
2. ip link set wlx00c0ca96611b down
3. iw dev wlx00c0ca96611b set type monitor

This adapter is confirmed working, so you got something wrong over there.
which branch are you using?

Muedi, 2018-08-06:
hi, 

I have the same issue with the AWUS036ACH on Kali in Virtualbox with a windows 10 host. 
I just cloned the repository again and reinstalled, got no errors during install.
first airodump works and "sniffs" perfectly, then the networkmanager chrashes (i suppose) and the wlan0 is not shown anymore in ifconfig.... 
restarted NM and wlan0 appeared to be functional. started airodump again, program runs, but shows no packages or devices...

any ideas? 
(i activated the monitormode maually like your side suggested and with ifconfig/ iwconfig instead, both show the same result)

kimocoder, 2019-02-15:
Network-Manager and other interference should be closed before operations in their domain.
If the problem persists, open this issue report again. Thank :+1:

closed 2019-02-15 by kimocoder

azamet90, 2019-10-23:
> Some of your steps there are.. just wrong.
> 
>     1. airmon-ng check kill
> 
>     2. ip link set wlx00c0ca96611b down
> 
>     3. iw dev wlx00c0ca96611b set type monitor
> 
> 
> This adapter is confirmed working, so you got something wrong over there.
> which branch are you using?

https://github.com/v1s1t0r1sh3r3/airgeddon/wiki/Consistent%20Network%20Device%20Naming
this in that link fix my wifi name problem you can try it