aircrack-ng/rtl8812au #158: 'net/rtl8812au' already registered

ZeroChaos-, 2018-08-05:
of note, both the 8812au.ko and 8814au.ko modules think they are rtl8812au as you can see from this paste.  note, this is from 8814au.ko and that 8812au.ko is *not loaded OR even installed"
```
[  481.096126] ------------[ cut here ]------------
[  481.096128] proc_dir_entry 'net/rtl8812au' already registered
[  481.096134] WARNING: CPU: 1 PID: 5628 at proc_register+0x9f/0x100
[  481.096134] Modules linked in: 8812au(O+) snd_hda_codec_hdmi intel_rapl x86_pkg_temp_thermal intel_powerclamp i915 kvm_intel kvm snd_hda_codec_realtek snd_hda_codec_generic irqbypass crct10dif_pclmul crc32_pclmul intel_cstate ofpart cmdlinepart mei_wdt intel_spi_platform intel_spi spi_nor iTCO_wdt iTCO_vendor_support drm_kms_helper snd_hda_intel 8814au(O) snd_hda_codec btusb drm btrtl snd_hda_core snd_hwdep btbcm btintel snd_pcm bluetooth fb_sys_fops syscopyarea sysfillrect sysimgblt snd_timer cfg80211 intel_uncore intel_gtt nct6775 intel_rapl_perf hwmon_vid e1000e snd coretemp i2c_i801 rfkill mei_me shpchp soundcore mei lpc_ich evdev ixgb ixgbe samsung_sxgbe tulip cxgb3 cxgb mdio cxgb4 vxge bonding vxlan macvlan vmxnet3 tg3 sky2 r8169 pcnet32 mii igb i2c_algo_bit dca e1000 bnx2 atl1c qemu_fw_cfg
[  481.096174]  cramfs mtd f2fs xfs jfs dm_verity reed_solomon dm_thin_pool dm_switch dm_snapshot dm_raid dm_log_writes dm_era dm_delay dm_cache_smq dm_cache dm_persistent_data dm_bio_prison firewire_core hid_sunplus hid_sony hid_samsung hid_pl hid_petalynx hid_monterey hid_microsoft hid_logitech_dj hid_logitech hid_gyration hid_ezkey hid_cypress hid_chicony hid_cherry hid_belkin hid_apple hid_a4tech sl811_hcd hid_generic mpt3sas aic94xx libsas lpfc nvmet_fc qla2xxx megaraid_sas megaraid aacraid sx8 DAC960 hpsa 3w_9xxx 3w_xxxx 3w_sas mptsas scsi_transport_sas mptfc mptspi mptscsih mptbase atp870u dc395x qla1280 imm parport dmx3191d sym53c8xx gdth advansys initio BusLogic arcmsr aic7xxx aic79xx sg pata_pcmcia pcmcia pcmcia_core nvme_fc nvme_loop nvmet nvme_rdma rdma_cm iw_cm ib_cm ib_core nvme_fabrics
[  481.096215]  virtio_net virtio_crypto crypto_engine virtio_mmio virtio_input virtio_balloon virtio_rng virtio_console virtio_scsi
[  481.096221] CPU: 1 PID: 5628 Comm: systemd-udevd Tainted: G        W  O    T 4.17.11-pentoo #1
[  481.096222] Hardware name:  /NUC5i5MYBE, BIOS MYBDWi5v.86A.0024.2015.0504.1120 05/04/2015
[  481.096224] RIP: 0010:proc_register+0x9f/0x100
[  481.096225] RSP: 0018:ffffbd60c18f3c28 EFLAGS: 00010286
[  481.096226] RAX: 0000000000000000 RBX: ffff9bcd38ad1478 RCX: 0000000000000006
[  481.096227] RDX: 0000000000000007 RSI: 0000000000000096 RDI: ffff9bcd5dc96390
[  481.096227] RBP: ffff9bcd38e5e820 R08: 0000000000000db6 R09: 0000000000000001
[  481.096228] R10: ffffe8ce4fdba740 R11: 0000000000000001 R12: ffff9bcd3bac5dd0
[  481.096229] R13: ffff9bcd3bac5e10 R14: ffff9bcd38e5e8a4 R15: ffff9bcd36be2660
[  481.096230] FS:  00007fd246db3800(0000) GS:ffff9bcd5dc80000(0000) knlGS:0000000000000000
[  481.096231] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  481.096232] CR2: 00007fd246ceef99 CR3: 00000003fad58001 CR4: 00000000003606e0
[  481.096233] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  481.096234] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  481.096235] Call Trace:
[  481.096239]  proc_mkdir_data+0x5e/0x90
[  481.096241]  ? 0xffffffffc10bf000
[  481.096268]  rtw_drv_proc_init+0x37/0xc9 [8812au]
[  481.096270]  ? 0xffffffffc10bf000
[  481.096280]  init_module+0x2d/0x1000 [8812au]
[  481.096283]  do_one_initcall+0x47/0x1a0
[  481.096285]  ? _cond_resched+0x10/0x40
[  481.096288]  ? kmem_cache_alloc+0x31/0x160
[  481.096290]  do_init_module+0x54/0x1d9
[  481.096291]  load_module+0x224c/0x2710
[  481.096294]  ? __se_sys_finit_module+0xac/0xc0
[  481.096295]  __se_sys_finit_module+0xac/0xc0
[  481.096297]  do_syscall_64+0x50/0xf0
[  481.096299]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  481.096300] RIP: 0033:0x7fd245a23eb9
[  481.096301] RSP: 002b:00007ffe306dffa8 EFLAGS: 00000202 ORIG_RAX: 0000000000000139
[  481.096302] RAX: ffffffffffffffda RBX: 0000634e418c9b60 RCX: 00007fd245a23eb9
[  481.096303] RDX: 0000000000000000 RSI: 00007fd2469bb732 RDI: 0000000000000006
[  481.096303] RBP: 0000000000000000 R08: 0000000000000000 R09: 0000000000000000
[  481.096304] R10: 0000000000000006 R11: 0000000000000202 R12: 0000000000020000
[  481.096305] R13: 0000634e4196e220 R14: 00007fd2469bb732 R15: 0000000000000000
[  481.096306] Code: e3 4c 8b 75 08 48 8b 73 a0 4c 89 f7 e8 6b fe ff ff 85 c0 78 e6 75 c9 49 8b 74 24 08 4c 89 f2 48 c7 c7 60 02 33 a2 e8 61 84 e3 ff <0f> 0b c6 05 d0 9f c5 01 00 8b 75 10 48 c7 c7 20 14 65 a2 81 c6 
[  481.096331] ---[ end trace 631700e340e51171 ]---
[  481.096362] WARNING: CPU: 1 PID: 5628 at rtw_drv_proc_init+0x56/0xc9 [8812au]
[  481.096363] Modules linked in: 8812au(O+) snd_hda_codec_hdmi intel_rapl x86_pkg_temp_thermal intel_powerclamp i915 kvm_intel kvm snd_hda_codec_realtek snd_hda_codec_generic irqbypass crct10dif_pclmul crc32_pclmul intel_cstate ofpart cmdlinepart mei_wdt intel_spi_platform intel_spi spi_nor iTCO_wdt iTCO_vendor_support drm_kms_helper snd_hda_intel 8814au(O) snd_hda_codec btusb drm btrtl snd_hda_core snd_hwdep btbcm btintel snd_pcm bluetooth fb_sys_fops syscopyarea sysfillrect sysimgblt snd_timer cfg80211 intel_uncore intel_gtt nct6775 intel_rapl_perf hwmon_vid e1000e snd coretemp i2c_i801 rfkill mei_me shpchp soundcore mei lpc_ich evdev ixgb ixgbe samsung_sxgbe tulip cxgb3 cxgb mdio cxgb4 vxge bonding vxlan macvlan vmxnet3 tg3 sky2 r8169 pcnet32 mii igb i2c_algo_bit dca e1000 bnx2 atl1c qemu_fw_cfg
[  481.096392]  cramfs mtd f2fs xfs jfs dm_verity reed_solomon dm_thin_pool dm_switch dm_snapshot dm_raid dm_log_writes dm_era dm_delay dm_cache_smq dm_cache dm_persistent_data dm_bio_prison firewire_core hid_sunplus hid_sony hid_samsung hid_pl hid_petalynx hid_monterey hid_microsoft hid_logitech_dj hid_logitech hid_gyration hid_ezkey hid_cypress hid_chicony hid_cherry hid_belkin hid_apple hid_a4tech sl811_hcd hid_generic mpt3sas aic94xx libsas lpfc nvmet_fc qla2xxx megaraid_sas megaraid aacraid sx8 DAC960 hpsa 3w_9xxx 3w_xxxx 3w_sas mptsas scsi_transport_sas mptfc mptspi mptscsih mptbase atp870u dc395x qla1280 imm parport dmx3191d sym53c8xx gdth advansys initio BusLogic arcmsr aic7xxx aic79xx sg pata_pcmcia pcmcia pcmcia_core nvme_fc nvme_loop nvmet nvme_rdma rdma_cm iw_cm ib_cm ib_core nvme_fabrics
[  481.096422]  virtio_net virtio_crypto crypto_engine virtio_mmio virtio_input virtio_balloon virtio_rng virtio_console virtio_scsi
[  481.096427] CPU: 1 PID: 5628 Comm: systemd-udevd Tainted: G        W  O    T 4.17.11-pentoo #1
[  481.096428] Hardware name:  /NUC5i5MYBE, BIOS MYBDWi5v.86A.0024.2015.0504.1120 05/04/2015
[  481.096449] RIP: 0010:rtw_drv_proc_init+0x56/0xc9 [8812au]
[  481.096450] RSP: 0018:ffffbd60c18f3c78 EFLAGS: 00010246
[  481.096451] RAX: 0000000000000000 RBX: ffffffffc16dce80 RCX: 00000000000000c1
[  481.096452] RDX: 00000000000000c0 RSI: ffff9bcd5dca7600 RDI: ffffe8ce4fe39780
[  481.096453] RBP: 0000000000000000 R08: 0000000000027600 R09: ffffffffa12a2d66
[  481.096454] R10: ffffe8ce4fdba740 R11: 0000000000000001 R12: ffffffffc16dcc80
[  481.096454] R13: ffffffffc17b4440 R14: ffffffffc17b4640 R15: ffff9bcd36be2660
[  481.096456] FS:  00007fd246db3800(0000) GS:ffff9bcd5dc80000(0000) knlGS:0000000000000000
[  481.096456] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  481.096457] CR2: 00007fd246ceef99 CR3: 00000003fad58001 CR4: 00000000003606e0
[  481.096458] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  481.096459] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  481.096459] Call Trace:
[  481.096461]  ? 0xffffffffc10bf000
[  481.096471]  init_module+0x2d/0x1000 [8812au]
[  481.096473]  do_one_initcall+0x47/0x1a0
[  481.096474]  ? _cond_resched+0x10/0x40
[  481.096476]  ? kmem_cache_alloc+0x31/0x160
[  481.096478]  do_init_module+0x54/0x1d9
[  481.096479]  load_module+0x224c/0x2710
[  481.096481]  ? __se_sys_finit_module+0xac/0xc0
[  481.096482]  __se_sys_finit_module+0xac/0xc0
[  481.096484]  do_syscall_64+0x50/0xf0
[  481.096486]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  481.096487] RIP: 0033:0x7fd245a23eb9
[  481.096488] RSP: 002b:00007ffe306dffa8 EFLAGS: 00000202 ORIG_RAX: 0000000000000139
[  481.096489] RAX: ffffffffffffffda RBX: 0000634e418c9b60 RCX: 00007fd245a23eb9
[  481.096489] RDX: 0000000000000000 RSI: 00007fd2469bb732 RDI: 0000000000000006
[... cut ...]
[  481.096491] R10: 0000000000000006 R11: 0000000000000202 R12: 0000000000020000
[  481.096492] R13: 0000634e4196e220 R14: 00007fd2469bb732 R15: 0000000000000000
[  481.096492] Code: 48 c7 c7 af 60 6e c1 e8 5e 08 c3 df 48 89 05 d7 7a 14 00 48 c7 c3 80 ce 6d c1 bd 00 00 00 00 49 c7 c4 80 cc 6d c1 48 85 c0 75 35 <0f> 0b eb bb 49 89 e8 48 c7 c1 80 cd 6d c1 48 8b 15 aa 7a 14 00 
[  481.096517] ---[ end trace 631700e340e51172 ]---
[  481.096526] Error: Driver 'rtl8812au' is already registered, aborting...
[  481.138068] ------------[ cut here ]------------
```

fariouche, 2018-08-05:
why does it think that it is already registered?? if the previous 8812au.ko is not even installed, the 8814au.ko should be the only one registering this name (which I agree is confusing)
I will modify the default registered name, but before that I would like to understand what happened here.
the 8814au.ko should be inserted only once.

kimocoder, 2018-08-05:
They both are showing rtl8812au in airmon-ng output, whilst the v5.1.5 branch shows rtl8812au for 8812/8821 and 8814au for 8814au

fariouche, 2018-08-05:
for me it make sense that the 8812au.ko is rejected, as the 8814au.ko already handles the 8812au.
If the driver was built with 8814 only, it will then report as 8814au I believe. (I'm still searching were this is done in the driver itself)

What I propose is to rename the driver to 88XXau.ko when both 14 and 12/21 are set.

fariouche, 2018-08-05:
This is set by the DRV_NAME define, hardcoded in autoconf.h

kimocoder, 2018-08-05:
@ZeroChaos- can you confirm if this issue has been resolved? If som, then just close it and we'll move along to the others :walking_man:

closed 2018-08-06 by kimocoder

ZeroChaos-, 2018-08-15:
building for 8814 and getting module 81xx seems to work fine for everything I tested.  I don't think I have an 8821 though.

kimocoder, 2018-08-15:
I've checked them all. They should be resolved :)
@fariouche got the idea and solved it masterly.

Left you a note on the aircrack-ng support commit for airmon-ng. Using airmon-ng with master breaks injection, but however, when running "aireplay-ng -9 wlan1" it switches to monitor and injection works.

ZeroChaos-, 2018-08-15:
if you mean adding 88xx then thanks! otherwise, I have no idea what you are talking about

kimocoder, 2018-08-15:
Yes that commit. It's great, but when using "airmon-ng start wlan1" the injection doesn't work for some reason, but when only using "aireplay-ng -9 wlan1" it's sets the adapters to monitor mode and injection is actually working..

ZeroChaos-, 2018-08-21:
both are doing the same thing.  I've tested over and over again and the only thing I can confirm is that sometimes everything works fine, and sometimes it doesn't work at all.  that includes tranistioning to monitor mode with and without airmon-ng; getting literally any packets in monitor mode; and functional injection.  Let's say stability of the 5.2.20 branch isn't "great"