aircrack-ng/rtl8812au #159: error building 8814a support when only 8814a is enabled

ZeroChaos-, 2018-08-05:
I simply set 8812a and 8821a to n and 8814a to y and it fails to build like this:
```
  gcc -Wp,-MD,/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/.rtw_debug.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -DCONFIG_CC_STACKPROTECTOR -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -DCC_HAVE_ASM_GOTO -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -funit-at-a-time -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_FXSAVEQ=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_CRC32=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_AVX512=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -pipe -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -DRETPOLINE -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-int-in-bool-context -O2 --param=allow-store-data-races=0 -fplugin=./scripts/gcc-plugins/structleak_plugin.so -fplugin=./scripts/gcc-plugins/randomize_layout_plugin.so -fplugin-arg-structleak_plugin-byref-all -DSTRUCTLEAK_PLUGIN -DRANDSTRUCT_PLUGIN -Wframe-larger-than=1024 -fstack-protector-strong -Wno-unused-but-set-variable -Wno-unused-const-variable -fomit-frame-pointer -fno-var-tracking-assignments -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fno-merge-all-constants -fmerge-constants -fno-stack-check -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -O1 -Wall -Wextra -Wno-unused-variable -Wno-unused-value -Wno-unused-label -Wno-unused-parameter -Wno-unused-function -Wno-unused -Wno-date-time -Wno-uninitialized -Wno-sign-compare -Wno-type-limits -Wno-date-time -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/include -DCONFIG_DISABLE_REGD_C -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/platform -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/hal/btc -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/hal/phydm -DCONFIG_MP_INCLUDED -DCONFIG_EFUSE_CONFIG_FILE -DEFUSE_MAP_PATH=\"/system/etc/wifi/wifi_efuse_.map\" -DWIFIMAC_PATH=\"/data/wifimac.txt\" -DCONFIG_LOAD_PHY_PARA_FROM_FILE -DREALTEK_CONFIG_PATH=\"/lib/firmware/\" -DCONFIG_TXPWR_BY_RATE_EN=0 -DCONFIG_TXPWR_LIMIT_EN=0 -DCONFIG_CALIBRATE_TX_POWER_TO_MAX -DCONFIG_RTW_ADAPTIVITY_EN=0 -DCONFIG_RTW_ADAPTIVITY_MODE=0 -DCONFIG_IEEE80211W -DCONFIG_BR_EXT '-DCONFIG_BR_EXT_BRNAME="'br0'"' -DCONFIG_WIFI_MONITOR -DCONFIG_RTW_NAPI -DCONFIG_RTW_GRO -DCONFIG_RTW_WIFI_HAL -DCONFIG_RTW_CFGVEDNOR_LLSTATS -DCONFIG_VHT_EXTRAS -DCONFIG_LED_CONTROL -DCONFIG_LED_ENABLE -DDM_ODM_SUPPORT_TYPE=0x04 -DCONFIG_LITTLE_ENDIAN -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT  -DMODULE  -DKBUILD_BASENAME='"rtw_debug"' -DKBUILD_MODNAME='"rtw_debug"' -c -o /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/.tmp_rtw_debug.o /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_debug.c
In file included from /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/include/drv_types.h:60:0,
                 from /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_debug.c:17:
/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_debug.c: In function ‘dump_drv_cfg’:
/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_debug.c:186:46: error: ‘MAX_RECVBUF_SZ’ undeclared (first use in this function); did you mean ‘MAX_XMITBUF_SZ’?
  RTW_PRINT_SEL(sel, "MAX_RECVBUF_SZ = %d\n", MAX_RECVBUF_SZ);
                                              ^
/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/include/rtw_debug.h:285:25: note: in definition of macro ‘RTW_PRINT_SEL’
    _seqdump(sel, fmt, ##arg) /*rtw_warn_on(1)*/; \
                         ^~~
/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_debug.c:186:46: note: each undeclared identifier is reported only once for each function it appears in
  RTW_PRINT_SEL(sel, "MAX_RECVBUF_SZ = %d\n", MAX_RECVBUF_SZ);
                                              ^
/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/include/rtw_debug.h:285:25: note: in definition of macro ‘RTW_PRINT_SEL’
    _seqdump(sel, fmt, ##arg) /*rtw_warn_on(1)*/; \
                         ^~~
make[2]: *** [scripts/Makefile.build:313: /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_debug.o] Error 1
make[1]: *** [Makefile:1571: _module_/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999] Error 2
make[1]: Leaving directory '/usr/src/linux-4.17.11-pentoo'
make: *** [Makefile:1903: modules] Error 2
```

kimocoder, 2018-08-05:
It's linked together.. "make RTL8814=1 && make install RTL8814=1"

kimocoder, 2018-08-05:
I close this as it is a duplicate of the other one

closed 2018-08-05 by kimocoder