aircrack-ng/rtl8812au #173: error building v5.2.20 on arm64 - unrecognized command line option ‘-mhard-float’

djjerdog, 2018-08-13:
DKMS make.log for rtl8812au-5.2.20.2 for kernel 4.17.14-sunxi64 (aarch64)
Mon Aug 13 10:10:00 CDT 2018
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.17.14-sunxi64/build M=/var/lib/dkms/rtl8812au/5.2.20.2/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.17.14-sunxi64'
  CC [M]  /var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_io.o
gcc: error: unrecognized command line option ‘-mhard-float’; did you mean ‘--warn-format’?
scripts/Makefile.build:312: recipe for target '/var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_cmd.o' failed
make[2]: *** [/var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
gcc: error: unrecognized command line option ‘-mhard-float’; did you mean ‘--warn-format’?
scripts/Makefile.build:312: recipe for target '/var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_security.o' failed
make[2]: *** [/var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_security.o] Error 1
gcc: error: unrecognized command line option ‘-mhard-float’; did you mean ‘--warn-format’?
scripts/Makefile.build:312: recipe for target '/var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_debug.o' failed
make[2]: *** [/var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_debug.o] Error 1
gcc: error: unrecognized command line option ‘-mhard-float’; did you mean ‘--warn-format’?
scripts/Makefile.build:312: recipe for target '/var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_io.o' failed
make[2]: *** [/var/lib/dkms/rtl8812au/5.2.20.2/build/core/rtw_io.o] Error 1
Makefile:1571: recipe for target '_module_/var/lib/dkms/rtl8812au/5.2.20.2/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl8812au/5.2.20.2/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.17.14-sunxi64'
Makefile:1928: recipe for target 'modules' failed
make: *** [modules] Error 2

kimocoder, 2018-08-21:
I'll take a look at this in few days. Thanks for reporting!

djjerdog, 2018-09-03:
I was able to get the driver to compile successfully by changing the following in **Makefile**

CONFIG_MP_VHT_HW_TX_MODE = y
to
CONFIG_MP_VHT_HW_TX_MODE = n

It looks like this change was made in commit [21b66c0c19c34d490db2a508e392214d059d8c0c](https://github.com/aircrack-ng/rtl8812au/commit/21b66c0c19c34d490db2a508e392214d059d8c0c)

kimocoder, 2018-09-03:
That's a great find. will push a commit in few days thanks!

kimocoder, 2018-09-03:
Pushed the changes to v5.2.20 and v5.3.4 repo. Thanks! Closing

closed 2018-09-03 by kimocoder