aircrack-ng/rtl8812au #174: Ubuntu 16.04 RTL8812AU Not connecting to Networks

viliuspovilaika, 2018-08-14:
Basically when I try to connect to a network with my Alfa AWUS036ACH using the RTL8812AU drivers on Ubuntu 16.04, after 5-10 seconds I am asked to reenter the password, which is 100% correct. Is there any troubleshooting for this?

christophfeinauer, 2018-08-16:
Same problem here:

```
wpa_supplicant -Dnl80211 -iwlx8026897216ac -c/home/XXX/wpa.conf
Successfully initialized wpa_supplicant

wlx8026897216ac: Trying to associate with 90:5c:44:7a:50:85 (SSID='XXX' freq=5220 MHz)
wlx8026897216ac: Associated with 90:5c:44:7a:50:85
wlx8026897216ac: CTRL-EVENT-DISCONNECTED bssid=90:5c:44:7a:50:85 reason=2
wlx8026897216ac: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
wlx8026897216ac: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="XXX" auth_failures=1 duration=10 reason=WRONG_KEY
wlx8026897216ac: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
```

I tried using wicd, disabling random mac, all branches in the repo and a different kernel (using `4.13.0-45-generic`).

howudodat, 2019-01-29:
+1 same problem with all 5G networks
```
Jan 29 03:04:53 localhost wpa_supplicant[3947]: wlan0: Trying to associate with 16:91:82:71:5b:b0 (SSID='NETWORK-guest' freq=5785 MHz)
Jan 29 03:04:53 localhost wpa_supplicant[3947]: wlan0: Association request to the driver failed
Jan 29 03:04:53 localhost wpa_supplicant[3947]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="NETWORK-guest" auth_failures=1 duration=10 reason=CONN_FAILED
Jan 29 03:05:03 localhost NetworkManager[1903]: <info>  [1548731103.1022] device (wlan0): supplicant interface state: disconnected -> scanning
Jan 29 03:05:03 localhost wpa_supplicant[3947]: wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="NETWORK-guest"
Jan 29 03:05:03 localhost wpa_supplicant[3947]: wlan0: Trying to associate with 16:91:82:71:5b:b0 (SSID='NETWORK-guest' freq=5785 MHz)
Jan 29 03:05:03 localhost wpa_supplicant[3947]: wlan0: Association request to the driver failed
Jan 29 03:05:03 localhost wpa_supplicant[3947]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="NETWORK-guest" auth_failures=2 duration=20 reason=CONN_FAILED
Jan 29 03:05:03 localhost NetworkManager[1903]: <info>  [1548731103.6771] device (wlan0): supplicant interface state: scanning -> disconnected
Jan 29 03:05:13 localhost NetworkManager[1903]: <info>  [1548731113.6546] device (wlan0): supplicant interface state: disconnected -> scanning
Jan 29 03:05:18 localhost NetworkManager[1903]: <warn>  [1548731118.1490] device (wlan0): Activation: (wifi) association took too long, failing activation
Jan 29 03:05:18 localhost NetworkManager[1903]: <info>  [1548731118.1491] device (wlan0): state change: config -> failed (reason 'ssid-not-found') [50 120 53]
Jan 29 03:05:18 localhost NetworkManager[1903]: <warn>  [1548731118.1506] device (wlan0): Activation: failed for connection 'NETWORK-guest'
Jan 29 03:05:18 localhost NetworkManager[1903]: <info>  [1548731118.1917] device (wlan0): supplicant interface state: scanning -> disconnected
Jan 29 03:05:18 localhost NetworkManager[1903]: <info>  [1548731118.1947] device (wlan0): state change: failed -> disconnected (reason 'none') [120 30 0]
```

howudodat, 2019-01-29:
For some reason I was able to connect one time (I think) but now still not able.  tried a reboot, but still no joy.  Here is the log:

```
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7116] keyfile: add connection in-memory (b50e2a69-9255-4e00-8f3a-ee9e11e266c1,"NETWORK_5GHz-guest")
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7164] device (wlan0): Activation: starting connection 'NETWORK_5GHz-guest' (b50e2a69-9255-4e00-8f3a-ee9e11e266c1)
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7195] keyfile: update /etc/NetworkManager/system-connections/NETWORK_5GHz-guest (b50e2a69-9255-4e00-8f3a-ee9e11e266c1,"NETWORK_5GHz-guest") and persist connection
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7199] audit: op="connection-add-activate" uuid="b50e2a69-9255-4e00-8f3a-ee9e11e266c1" name="NETWORK_5GHz-guest" pid=3810 uid=0 result="success"
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7237] device (wlan0): state change: disconnected -> prepare (reason 'none') [30 40 0]
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7275] device (wlan0): state change: prepare -> config (reason 'none') [40 50 0]
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7285] device (wlan0): Activation: (wifi) connection 'NETWORK_5GHz-guest' requires no security.  No secrets needed.
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7288] Config: added 'ssid' value 'NETWORK_5GHz-guest'
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7289] Config: added 'scan_ssid' value '1'
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7289] Config: added 'key_mgmt' value 'NONE'
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7331] sup-iface[0x2d389b80,wlan0]: config: set interface ap_scan to 1
Jan 29 15:44:40 localhost NetworkManager[1898]: <info>  [1548776680.7477] device (wlan0): supplicant interface state: disconnected -> scanning
Jan 29 15:44:46 localhost wpa_supplicant[2177]: wlan0: Trying to associate with 16:91:82:71:5b:b0 (SSID='NETWORK_5GHz-guest' freq=5785 MHz)
Jan 29 15:44:46 localhost wpa_supplicant[2177]: wlan0: Association request to the driver failed
Jan 29 15:44:46 localhost wpa_supplicant[2177]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="NETWORK_5GHz-guest" auth_failures=1 duration=10 reason=CONN_FAILED
Jan 29 15:44:46 localhost NetworkManager[1898]: <info>  [1548776686.8566] device (wlan0): supplicant interface state: scanning -> disconnected
Jan 29 15:44:56 localhost NetworkManager[1898]: <info>  [1548776696.8508] device (wlan0): supplicant interface state: disconnected -> scanning
Jan 29 15:44:56 localhost rsyslogd-2007: action 'action 9' suspended, next retry is Tue Jan 29 15:45:56 2019 [v8.16.0 try http://www.rsyslog.com/e/2007 ]
Jan 29 15:44:57 localhost wpa_supplicant[2177]: wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="NETWORK_5GHz-guest"
Jan 29 15:44:57 localhost wpa_supplicant[2177]: wlan0: Trying to associate with 16:91:82:71:5b:b0 (SSID='NETWORK_5GHz-guest' freq=5785 MHz)
Jan 29 15:44:57 localhost wpa_supplicant[2177]: wlan0: Association request to the driver failed
Jan 29 15:44:57 localhost wpa_supplicant[2177]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="NETWORK_5GHz-guest" auth_failures=2 duration=20 reason=CONN_FAILED
Jan 29 15:44:57 localhost NetworkManager[1898]: <info>  [1548776697.4179] device (wlan0): supplicant interface state: scanning -> disconnected
Jan 29 15:45:01 localhost ntpd[2216]: 46.175.224.7 local addr 172.16.4.50 -> <null>
Jan 29 15:45:06 localhost NetworkManager[1898]: <warn>  [1548776706.0138] device (wlan0): Activation: (wifi) association took too long, failing activation
Jan 29 15:45:06 localhost NetworkManager[1898]: <info>  [1548776706.0139] device (wlan0): state change: config -> failed (reason 'ssid-not-found') [50 120 53]
Jan 29 15:45:06 localhost NetworkManager[1898]: <warn>  [1548776706.0154] device (wlan0): Activation: failed for connection 'NETWORK_5GHz-guest'
Jan 29 15:45:06 localhost NetworkManager[1898]: <info>  [1548776706.0317] device (wlan0): state change: failed -> disconnected (reason 'none') [120 30 0]
Jan 29 15:45:06 localhost kernel: [  981.114553] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Jan 29 15:45:06 localhost NetworkManager[1898]: <info>  [1548776706.6064] device (wlan0): supplicant interface state: disconnected -> inactive
Jan 29 15:45:07 localhost ntpd[2216]: 138.68.46.177 local addr 172.16.4.50 -> <null>
```

howudodat, 2019-01-29:
ok, I have found a pattern.  I can connect to 2.4GHz and 5GHz networks, however I have to either remove the driver and reload it or reboot between connections.  The below set of commands will connect to 2G, then disconnect, then connect to 5G

```
nmcli con add type wifi ifname wlan0 con-name 'NETWORK' ssid 'NETWORK'
nmcli con modify 'NETWORK' wifi-sec.key-mgmt wpa-psk; nmcli con modify 'NETWORK' wifi-sec.psk '********';
nmcli con up id 'NETWORK'
-----------------  WE ARE CONNECTED -------------------------------

nmcli con down NETWORK; nmcli con delete NETWORK
modprobe -r 88XXau && modprobe 88XXau

nmcli con add type wifi ifname wlan0 con-name 'NETWORK5' ssid 'NETWORK5'
nmcli con modify 'NETWORK5' wifi-sec.key-mgmt wpa-psk; nmcli con modify 'NETWORK5' wifi-sec.psk '********';
nmcli con up id 'NETWORK5'
-----------------  WE ARE CONNECTED -------------------------------

nmcli con down NETWORK5; nmcli con delete NETWORK5
modprobe -r 88XXau && modprobe 88XXau
```

haxxinen, 2019-02-12:
This worked for me to get a stable connection on 5GHz:
```
# /etc/init.d/networking stop
# service NetworkManager stop
# modprobe -r 88XXau && modprobe 88XXau # or disconnect/connect device
# wpa_supplicant -i wlan0 -c <(wpa_passphrase <SSID> <PSK>)
# dhclient wlan0
```

Doppel-D, 2019-09-13:
Same here, but worse ;- ) Never got a connection

[Similar thread](https://github.com/aircrack-ng/rtl8812au/issues/115#issuecomment-531158542)

aldenq, 2019-11-06:
same exact problem as OP. 
output of uname:
Linux aldenq-XPS-13-9360 4.15.0-66-generic #75~16.04.1-Ubuntu SMP Tue Oct 1 14:01:08 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

and i am using the 5.6.4.1 version of the driver. 

if it means anything, i checked wireshark during the EAPOL handshake and it looks like only 2 of the 4 EAPOL packets are getting sent.

aldenq, 2019-11-07:
ok, found a solution, use the 5.6.4.2 branch rather than the 5.6.4.1 branch

kimocoder, 2019-11-14:
Issue should be resolved on branch v5.6.4.2. Please update.
If any other issues is noticed, feel free to open a new issue and let us know. Thanks!

closed 2019-11-14 by kimocoder