aircrack-ng/rtl8812au #177: Ubuntu 16.04 RTL8812AU Multiple Issues

viliuspovilaika, 2018-08-21:
To make it short, after using these drivers for a week I've noticed multiple major issues when using them with my Alfa AWUS036ACH:

1. The connection drops randomly, mostly on heavy use (in example when running speedtest-cli)
2. The signal strength is horrible, being right away from the router I get -55 dBm (also, is that supposed to be a negative number?)
3. There is literally no 5Ghz support. When I try to switch to 5.**G frequencies or at all, any other frequency (including the other 2,4Ghz frequencies) than the default on iwconfig, I get zero results.

Is there a roadmap by chance for the drivers? Are these issues being worked on? Does anyone else encounter them by chance?

Just letting you know, I'm running Ubuntu 16.04 on 4.15.0-32-generic

kimocoder, 2019-02-15:
Old news.. this was a bad, hardcoded, mixed old/new code release from Realtek.
But it's getting better for every release and fixes that gets added.

Now we're moving from v5.2.20(.2) over to v5.3.4 in a month or so (officially), I run it now.
A bit more stable, less ping buffer, more additions and functions.

And we've been doing lots the past 2 years, I mean alot.. the best would have been to just start from scratch and write/make a mac80211 driver instead of the mixed nl80211/cfg80211 with lots of functions missing, while the rest was hardcoded :+1:

kimocoder, 2019-11-14:
Update to a newer branch, preferred is the v5.6.4.2. Closing this issue

closed 2019-11-14 by kimocoder