aircrack-ng/rtl8812au #178: Error!! No TCP Packets Sniffed under Monitor mode

spaul13, 2018-08-22:
I am using the RTL8812AU external USB wifi adapter in the monitor mode using the driver code you guys provided and following the steps to open the monitor mode. But still, I can't sniff any TCP data packets using the tcpdump. But I can sniff the UDP packets. Can anyone (@kimocoder, @aircrack-ng) please tell me why the TCP data packets can't be sniffed? 

Is there any configurations that I have to enable to make things happen, please tell.

ValZapod, 2019-02-15:
@kimocoder That is bollocks, I suppose. Transport level is too far to have problems with monitor mode... Though there may be problems with udp over multicast which is processed by driver separately... Or smth like that. But I still think it is not possible...

James-stdy-app, 2020-01-13:
Bump. Is there any news on this or any thing that worked to capture TCP packets. Have the same issue

ValZapod, 2020-01-13:
Can you give wireshark screenshots?

James-stdy-app, 2020-01-13:
Have not tested in wireshark, however when using the Scapy python library with the filter=“tcp” thing set there are no packets coming through. Or even without the filter no packets with a TCP later are found :/

ValZapod, 2020-01-13:
@James-stdy-app Test in wireshark or tcpdump.

James-stdy-app, 2020-01-13:
Ok will do. Will report back shortly

James-stdy-app, 2020-01-13:
[image]
Tested using TCPDUMP the command I ran is on the bottom. I have never used TCPDUMP before. So I presume if I am picking up packets then it’s NOT a driver issue? Is this correct

ValZapod, 2020-01-13:
OMG. Rasberry Pi.
 use `sudo tcpdump -i wlan1 tcp` if there are no packets, give me sudo tcpdump -i wlan1 -w tcppackets.pcap  You need to open it in wireshark.

James-stdy-app, 2020-01-13:
Thank you very much for your help. Yeah sorry probably should have mentioned it was run on a raspberry pi device. Unfortunatley there doesn’t seem to be any packets at all, thus giving an empty “tcppackets.pcap” file. Any ideas?

ValZapod, 2020-01-13:
2 packets recieved, run without -w option but with -vv option.

ValZapod, 2020-01-13:
Or better use tcpdump -i wlan1 -nnvvXSs 1514 tcp

James-stdy-app, 2020-01-13:
Yeah same thing :/

As you can see card is in monitor mode.
The card is a Alfa AWUS036ACS 
Which is running on v5.6.4.2 driver.
I saw that v5.3.4 is the most stable, maybe I should try that?
[image]

ValZapod, 2020-01-13:
Okay. So do you understand that those packets are encoded, right? WPA2. You will not see tcp packets untill you will decode them. Do you connected to any wlan? If yes then you will see those packets. If not, well... Also consider this. This is so called radiotap https://superuser.com/questions/915250/tcpdump-not-displaying-rssi layer. It is not really parsable by tcpdump. Maybe some latest versions... So you can go to some wifi hotspot or youcp can put you router on without password mode.

James-stdy-app, 2020-01-13:
Hey sorry I’m a total network noob, trying to learn as much as I can :) . Again thanks so much for your help. Yes I am connected to a WPA2 network, is decoding an easy task?

James-stdy-app, 2020-01-13:
I really only want to get the IP address from these packets is that encrypted too?

ValZapod, 2020-01-13:
Ip addresses are encrypted too. Obviously. Only mac addresses of devices are visable. https://serverfault.com/questions/587406/which-layer-of-the-osi-model-is-responsible-for-encryption-in-wlans

Ip addresses are not unique. Everubody has 192.168.1.0 as a router for example.

ValZapod, 2020-01-13:
If you have captured 4 EAPOL files of a hadshake, then (when you know the password or you need to bruteforce it) you can decode the transmission in wireshark.

ValZapod, 2020-01-13:
@James-stdy-app https://youtu.be/G_O_NeNAVuU see the video.

kimocoder, 2020-01-20:
Have pushed a fix for pcap_activate, I've seen some issues that that issue is affiliated with scapy.
This fix isn't pushed to Kali, so the driver has to be built from this repo and tested.

ValZapod, 2020-01-20:
@kimocoder You are fast)) BTW, next time you will talk to Realtek can you mention this https://github.com/aircrack-ng/rtl8812au/issues/450#issuecomment-545790731
It is driving me mad ;) Or maybe you know how to fix, or Microsoft can fix it.