aircrack-ng/rtl8812au #179: New xhipset support request: rtl8188eus

kimocoder, 2018-08-24:
From:
https://github.com/quickreflex/rtl8188eus/tree/v5.2.2.4

More info tba

kimocoder, 2018-08-24:
Could this be done @fariouche ? It would possibly require same modifications as the 8814au I would think?

fariouche, 2018-08-24:
It should not be too hard... but there is no changelog, so It's difficult to see if the 5.2.2.4 is very different from the 5.2.2.0 we have... I'm checking

kimocoder, 2018-08-24:
True. There are 2 branches available over there, we only need the HAL. will take a look at it this weekend

fariouche, 2018-08-24:
Took a quick look... two difficulties so far:
- missing rtl8188e/HalEfuseMask8188E_USB.h. In fact all the efuse folder is missing and I'm still unsure about an equivalent in the tree...
- Some EFUSE define are clashing with the ones from 8812a... Especially one, that have a different value: BANK_NUM value is 3 in 8188, 2 in 88xx. (used as EFUSE_BT_REAL_CONTENT_LEN)

The first is blocking me for now.
I will not be able to look at it any further until mid-september, going to hollidays

kimocoder, 2018-08-24:
Ok. The simpliest would be to leave it standalone. I could easily add injection support to it as-is anyway. Leaving this here anyways. Have a great holiday sir! Best regards

kimocoder, 2018-09-03:
We keep this separated. Closing

closed 2018-09-03 by kimocoder