aircrack-ng/rtl8812au #180: NEW version v5.3.4

kimocoder, 2018-08-24:
We've found a new release from Edimax. It's versioned 5.3.4.
What's interesting here, it actually got injection working allready.

The driver only support rtl8821au atm.
Work started.

kimocoder, 2018-08-24:
@fariouche I see some interesting things in the code.. I think they have used some code found around + added own stuff on top on that again. The driver actually support injection allready, I bet I know where that came from ;)

The driver looks promising on several places. Got more time to look at it soon.

fariouche, 2018-08-24:
cool. Agree with you, same feeling, better base here.
Do not forget that until now, we used various patches coming all from various older versions of realtek. So it's not surprising me that we found them all here.
I see that the main new feature is mesh, don't know what use will be for us.

I will need to add the multi chip support again I think

kimocoder, 2018-08-25:
Yes, step 1 should be to get all 3 chipset working, from there I have to do a walkthrough on whats added and not, as there is no changelog or commit log available.

fariouche, 2018-08-25:
@kimocoder , I've pushed you my 5.3.4 branch, with the 8814 support in it :-)
It still needs some work. In AP mode I'm missing RX_STBC1 option... I doubt that the VHT mode is enabled for the 8814 at least.
I see also a kernel oops when rmmod the driver sometimes, to be investigated.

Injection may not work..

But globaly it works. I've tested the 8821 or the 8811 yet. But this should be a good base for you to continue until I return mid-september

kimocoder, 2018-08-25:
I see it, there are conflicts to the PR, anyway I got lots to do with it besides the chip support. :)

kimocoder, 2018-08-25:
I'll go ahead and try it see how it actually works out. Just leave the PR open I'll grab it after finner. Thanks!

fariouche, 2018-08-25:
the oops seems to be linked to this function:
_rtw_hal_macid_bmp_sleep

I'm not sure why, but they changed the function code. Maybe a pointer setting is missing in the 8814 specific part. Should not be too hard to fix

kimocoder, 2018-08-25:
This build is actually ported from the master (Larry Finger aka lwfinger) at https://github.com/lwfinger I've found. That's also great need, cause his the best at Realtek there is. Great news, I started working on unreachable switches and compile errors. Hard to see whats done and not.. since there is no commit log, I have to do a major walkthrough

fariouche, 2018-08-25:
I checked the lwfinger repo, lot of drivers, but they seem all very old code base.

Indeed, I see also many compilation warnings that may hide real bugs...

FYI, I just fixed the bug for the 8814. not more warnings/oops
I've put values in the hal that I believe correct, but not 100% sure

fariouche, 2018-08-25:
this will be my last commit. I will keep reading my email from time to time

kimocoder, 2018-08-25:
Just pushed the changes manually. Fixed USB3 with the modeswitch and compilation warnings. Had to throw in you 8814 kernel oops fix by hand, cause your repo can't merge with both mine or aircrack-ng, so when you get back from holiday, delete your fork and re-fork it to get it to work again. Have a GREAT holiday sir! :1st_place_medal: :+1:

kimocoder, 2018-09-03:
Yup, thanks for the heads up. Using the above comment as a TODO. putting it in there

fariouche, 2018-09-12:
"txpower control doesn't seem to work anymore (worked with v5.2.20(.2))"
-> I do remember having seen something about it in the code, and I didn't include them. Should be easy to fix

kimocoder, 2018-09-12:
Yes the working commit for the v5.2.20 branch is here https://github.com/aircrack-ng/rtl8812au/commit/f6461d311d5732a340b1381899818f4729b553f6

I'm moving and starting on a new job shortly and got to spend some time getting in place over there, but will be back soon again.

fariouche, 2018-09-16:
the txpower commit is already in the 5.3.4...
Is the twpower control not working for all chipsets? or only the 8814 has the issue?

kimocoder, 2018-09-16:
I think it's all, not sure at all but

kimocoder, 2018-09-22:
@Mister_X may you change "standard branch" om settings, instead of "v5.1.5" branch, the "v5.2.20" may be set instead

francoism90, 2018-09-23:
@kimocoder Is it correct you used EW-7811UAC for the source? :)
And I'm a bit confused; what branch should I use for testing this one?

kimocoder, 2018-09-23:
Correct, Edimax sources on the v5.3.4 with all patches from v5.2.20(.2) on top of it

francoism90, 2018-09-23:
@kimocoder Great work! I'm happy to test and debug if needed. :)

fariouche, 2018-09-24:
For info, I'm currently working with the 8814 to improve speed inAP mode.
I'm currently stuck at 300mbps, even if the iw list displays 1300mbps max.
My phone can go up to 866Mbps.

@kimocoder, IAsked realtek for source code, no answer yet. I think that realtek will share a driver as is to alpha if they request it. if we can get a hand on the 5.3.4 with 8814, it will help a lot.

kimocoder, 2018-09-24:
Ok. I'm still a bit busy with moving, but will be back online again soon.

francoism90, 2018-09-24:
@fariouche Same here, it says 1300mbps, also reaching a max. speed of around 300mbps.
Glad you're looking into tweaking it. :)

In the past I've also contacted Realtek for the source code of the 8814au.. haven't received any response in months, so don't expect a reply. :/

fariouche, 2018-09-24:
no problems, it is not urgent.

francoism90, 2018-09-25:
I'm unable to set monitor mode, is this a known bug? It results in a module error.
To restore I need to unload and reload the 88xx module.

Please let me know if you need any debug info. :)

fariouche, 2018-09-25:
Nerver tested the monitor mode.
Can you tell me the commands you used to enter monitor mode?
Can you also post the dmesg output if it crashed ?(if any)

For the tx power of the 8814, I'm still looking at it, but maybe a bug in the power management. I'll test a fix in a few hours

fariouche, 2018-09-25:
About the speed of the 8814, seeing the following table (https://d2cpnw0u24fjm4.cloudfront.net/wp-content/uploads/80211-Data-rates-and-snr-requriements-screen-shot.png) It seems that MCS8 and MCS9 are not selected.
300Mbps means Short GI-40MHz VHT MCS7 2SS. My phone is 2 SS so it is expected, but I do not understand why it didn't select MCS9...
For the 80MHz bandwidth, I still need to check if my hostapd.conf file is using a compatible channel (I doubt that) so it may explain that I do not have 80Mhz.

fariouche, 2018-09-28:
For information about 8814:

- [ ] 8814au 5ghz got issues, while for 8812/8821 it's working. --> I cannot reproduce the issue
- [x] 8814au VHT got issues, report in below comment --> VHT is working, but limited, see below
- [ ] Cannot set channel with iw/netlink (v5.1.5 and v5.2.20 works)
- [ ] txpower control doesn't seem to work anymore (worked with v5.2.20(.2)) --> indeed, iw is not even calling the set_tx_power... I have to dig inside iw to understand the condition.
- [ ] the get channel function got a compile error related to "CHANNEL_WIDTH_5" and "CHANNEL_WIDTH_10" this was working properly in v5.1.5 again.
- [ ] Cannot negociate above 300Mbps when in AP mode. I have the exact same issue with the 8812 and 8814 so this is not related to the 8814 code... The code should work, so maybe a hostapd issue? Still investigating...
- [ ] No LEDs with 8814. Not sure why, all the functions are here and doing the same as before... I will try with an older driver to see if the led is working

Lot of work to unleash the full potential of theses chips

francoism90, 2018-09-30:
@fariouche Thanks for looking into the issues and trying to solve them. Such a shame Realtek isn't able to produce a good working driver. :/

I can confirm it seems to be capt at 300Mbps max. (5Ghz, 802.11ac, ASUS AC-78U), the same issue (with same device) happens on Windows 10 though, so maybe something is happening with the Realtek-device itself?
LED indeed doesn't work, but I'm happy because the one on the AC-68U-USB simple blinds you. ;)

Thanks again and if debug/testing is needed, please let me know. :)

fariouche, 2018-09-30:
May have found the reason why VHT is not enabled (for all chipsets!)
Got a patch, that needs testing, but I got my phone to negociate the link at 780Mbps. (this means 80Mhz MCS8, almost the max of my phone. MCS9 is maybe not selected because of too many wifi around)

THe problem is a bad condition with MESH mode... why? I do not know.
Eitther the condition is wrong (and commenting the condition makes it work) or the MESH mode is wrongly reported.
I had the feeling that the mesh mode is not yet fully supported by this driver (because it actually appeared with this driver!)

Keep you in touch as soon as I've make a deeper look and more tests about the actual speed

fariouche, 2018-10-04:
Until kimocoder got time to commit the VHT fix here, I've looked at the tx power issue.
iwconfig is not supported. The ioctl is not even coded in the driver. So forget it (iwconfig is deprecated anyway)
Using iw, you have to be careful. The tx power unit is milli-Bel-milliwatt.

Some info on the internet shows the wrong command, that may work for kernel < 2.6.36 or with the option COMPAT_KERNEL_RELEASE.

I got it tx power working using the following command line:
sudo iw dev wlan0 set txpower fixed 1100

and not 11mBm as shown in some pages. So, do not forget to multiply the value by 100.

After that, iw dev wlan0 info display the correct txpower.

Now, I have no way to verify that the txpower is actually applied. The driver will store the value in a variable, call a lower level function to actually set the tx power and... well no errors, but not sure either.
Calling the get txpower ioctl will just return the value stored previously. Seems to be no way to actually probe the firmware for the actual really applied value....

If someone is able to verify, it will be great

kimocoder, 2018-10-04:
Ok. I've merged the VHT fix. iwconfig is deprecated too, so iw should be used in the future.
I'm back on ny computer again in 4 days now, finally

francoism90, 2018-10-05:
@kimocoder Thanks for the merge! Do you need debug info?
So far downloading seems to be working fine ~28.8MB/s pretty stable. :)

fariouche, 2018-10-07:
testing the wifi at 866Mbps I can reach the maximum real bandwith (270Mbps). Really good.
Also fixed the led blinking.
This can be disabled by commenting the CONFIG_RTW_LED define in autoconf.h

francoism90, 2018-10-08:
@kimocoder @fariouche I'm really impressed! Finally I'm able to browse and download without issues. :)
Will try to test the LED.

fariouche, 2018-10-08:
I'm essentially using it as an AP, so glad it works good in station mode.
I believe that v5.3.4 is the best driver we have so far. Still need some corner fixes.
Monitor and injection mode need to be confirmed yet.

francoism90, 2018-10-08:
Just some build errors on latest commit:
```
DKMS make.log for rtl8812au-5.3.4 for kernel 4.18.12-1-ck-skylake (x86_64)
ma  8 okt 2018 15:58:53 CEST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.18.12-1-ck-skylake/build M=/var/lib/dkms/rtl8812au/5.3.4/build  modules
make[1]: Entering directory '/usr/lib/modules/4.18.12-1-ck-skylake/build'
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/wifi_regd.o
/var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_get_channel’:
/var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_cfg80211.c:5563:5: warning: enumeration value ‘CHANNEL_WIDTH_5’ not handled in switch [-Wswitch]
     switch(pHalData->current_channel_bw){
     ^~~~~~
/var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_cfg80211.c:5563:5: warning: enumeration value ‘CHANNEL_WIDTH_10’ not handled in switch [-Wswitch]
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_mp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_intf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_com.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_com_phycfg.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_phy.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_dm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_dm_acs.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_mp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_mcc.o
[... 14 lines cut ...]
/var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_mp.c:2091:8: note: in expansion of macro ‘IS_HARDWARE_TYPE_8814A’
    if (IS_HARDWARE_TYPE_8814A(padapter))
        ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_mp.c:2091:31: note: each undeclared identifier is reported only once for each function it appears in
    if (IS_HARDWARE_TYPE_8814A(padapter))
                               ^~~~~~~~
/var/lib/dkms/rtl8812au/5.3.4/build/include/drv_types.h:1616:46: note: in definition of macro ‘rtw_get_hw_type’
 #define rtw_get_hw_type(adapter) (((PADAPTER)adapter)->dvobj->HardwareType)
                                              ^~~~~~~
[... cut ...]
/var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_hci/hal_usb.c: In function ‘usb_init_recv_priv’:
/var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_hci/hal_usb.c:28:8: warning: cast between incompatible function types from ‘void (*)(void *)’ to ‘void (*)(long unsigned int)’ [-Wcast-function-type]
        (void(*)(unsigned long))usb_recv_tasklet,
        ^
make[1]: *** [Makefile:1521: _module_/var/lib/dkms/rtl8812au/5.3.4/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.18.12-1-ck-skylake/build'
make: *** [Makefile:2054: modules] Error 2
```

fariouche, 2018-10-08:
oops, I was sure to have tested... sorry. Fixed. Should be available shortly

fariouche, 2018-10-09:
Status information about 8814:

* [x]  8814au 5ghz got issues, while for 8812/8821 it's working. --> I cannot reproduce the issue
* [x]  8814au VHT got issues, report in below comment --> VHT is working, but limited, see below
* [ ]  Cannot set channel with iw/netlink (v5.1.5 and v5.2.20 works)
* [x]  txpower control doesn't seem to work anymore (worked with v5.2.20(.2)) --> indeed, iw is not even calling the set_tx_power... I have to dig inside iw to understand the condition.
* [ ]  the get channel function got a compile error related to "CHANNEL_WIDTH_5" and "CHANNEL_WIDTH_10" this was working properly in v5.1.5 again.
* [x]  Cannot negociate above 300Mbps when in AP mode. I have the exact same issue with the 8812 and 8814 so this is not related to the 8814 code... The code should work, so maybe a hostapd issue? Still investigating...
* [x]  No LEDs with 8814. Not sure why, all the functions are here and doing the same as before... I will try with an older driver to see if the led is working
* [x] Fixed 8811 and 8821 support

francoism90, 2018-10-09:
@fariouche How can I disable the LED? :)
I have changed the `include/autoconf.h` line to `/*#define CONFIG_RTW_LED*/`.

fariouche, 2018-10-09:
indeed, it does not compile. Will fix it asap

fariouche, 2018-10-09:
By the way, after looking at the code (current and old ones, like 4.3.21) I do not see how the set_txpower can work.
The ioctl only set a variable, and return it when the get_txpower is return. But internally it is totally ignored.
So a really doubt that the txpower is actually working at all. It's all fake.!

fariouche, 2018-10-09:
@francoism90 , you will be able to disable leds by commenting CONFIG_RTW_SW_LED as soon as the fix is pushed here

kimocoder, 2018-10-10:
Pushed. Awesome, you really did some magic over there @fariouche

fariouche, 2018-10-11:
thanks ;-) . Still not happy myself. I would like to push the tx power... But one comment in the code by realtek makes me fear to destroy the dongle. (something about limiting to some kind of magic value that they got by experimenting because they don't want to fry the power module...)

kimocoder, 2018-10-11:
Remember my dear frenchie, you could always do it, suggested adapter would be the 8811 chipset.

You know our favorite manufacturer would sacrifice some hardware to fix the most important issues of them all, the one and beloved txpower (range/distance) I'm pretty sure it's allright, just be sure to probe through the code before frying 'em all 😅

francoism90, 2018-10-14:
@fariouche Hmm, I'm still getting build errors with `CONFIG_RTW_SW_LED` commented in `include/autoconf.h`. Are you sure this is the correct line? :)

fariouche, 2018-10-14:
Yes I'm sure. But I just check the code here and it is not correct. My repo and kimo's one is correct... so something wasn't pushed here completely. @kimocoder ?

kimocoder, 2018-10-14:
I actually forgot to pull it. Try again, I just did so

kimocoder, 2018-10-18:
No problem, we're happy to resolve the issue.
@fariouche I got a good idea.. found a Realtek driver that actually works like a charm, and that's the rtl8187 driver.. starting inspection of it, 'cause ut doesn't have the issues we're struggling with and the range (txpower) is AMAZING on it.

fariouche, 2018-10-18:
I'm currently fighting with the tx power... I've read all the code, and implemented a fix, but not sure about testing. If I set the 8811 as an AP using hostapd, change the txpower, I'm expecting to have a big impact on the range... I can see a difference, but not very sure.

Where is that rtl8187 driver? github?

kimocoder, 2018-10-18:
Information regarding rtl8187 is [herr](https://www.aircrack-ng.org/doku.php?id=r8187)

Do you need a USB ammeter for proper testing? I got the ultimate USB ammeters with bluetooth support and APP for it. I'll ship you one!

rtl8187 (r8187) has different versions, both ieee80211 and a mac80211 (which I think is included upstream debian maibtained by Johannes Berg) *my thought without confirmation* er have a ieee80211 driver here, som why not start there. Anyways, the Kali version or it kicks ass, my old Alfa adapter with the chipset is really SUPERIOR what matters range and penetration testing usages (mon/inject support) working flawless

fariouche, 2018-10-18:
found some code in github... it is rather old (5years).
This chip seems different too. Everything is managed by the driver, with power tables etc.
What makes things difficult with the 8812/21 etc is that the tx power is managed by the firmware, and we do not have access to the tables.

The driver can only write a tx power index in a register, and the firmware will decide what to do with it. What is the corresponding dBm value? impossible to say (and seeing some realtek comments, they don't too...). This is just a matter of calibrating the values. WIth a spectrum analyzer, it may be possible to get real dBm values.

To make things even worse, some dongles are flashed with a country code that cannot be changed. They do not allow any tx power enforcement :-( 

As far as I understood the driver's logic, at some point, settcpower is called.
The function is then getting the proper tx power index depending on various tables (loaded from the eeprom) and current link settings (bandwith, rate, channel etc)
So, theorically, we should be able to fix the tx power by alway returning a fixed tx power index.

I do not have an usb ampmeter, this is indeed a good idea.
I've looked at the ieee80211 version of the driver, found on github.

I can have access to a spectrum analyzer at work, but the access it limited, so I'm authorized only when not in use and only for 15min. I must be sure and quick lol.

kimocoder, 2018-10-18:
Ok. Integrering! Still not got my computer by hand, but hopefully try going getting ny hands on it this weekend again :+1: A USB ammater is on it's way!

kimocoder, 2018-10-20:
Just pushed various fixes / cleanups.

* Remove TESTCHIP support
* Nuke rtw_proc aka debug via /proc
* Minor fix on the inconsistent teardown (from upstream)
* Hard disable old debug read/write file functions
* Remove unneeded read firmware support
* Remove unneeded objs from Makefile 
* Remove more objs from unsupported chips
* Makefile cleanup
* Remove unused firmware load
* Support kernel PIC mode for compiling

francoism90, 2018-10-21:
@kimocoder Going to test and let you know, thanks for your hard work!
Does this also mean this fixes the TX-power issues?

kimocoder, 2018-10-21:
No it removes some firmware load mostly, besides that it looks more cleaner and easier for devs to probe through

Yes21, 2018-11-05:
Does 5.3.4 compiles on RPI ?

I can't see CONFIG_PLATFORM_ARM_RPI in the 
makefile ...

fariouche, 2018-11-06:
should be.
This flag does not make sense at the first place, this is a normal linux after all

ohquait, 2018-11-07:
If I need the rtl8821au, how can I build the one in 5.3.4?
Do I have to change the makefile in CONFIG_BT_COEXIST = y?
Because the 8821au chips come together with BT.

[12 more replies]

closed 2019-02-15 by kimocoder