aircrack-ng/rtl8812au #181: Raspian Prob.

exotic-fannypack, 2018-08-25:
Driver isn't building on rpi3 raspbian latest. Will try again after fresh rasp install. 
```
GNU nano 2.7.4  File: /var/lib/dkms/rtl8812au/5.1.5/build/make.log            

DKMS make.log for rtl8812au-5.1.5 for kernel 4.14.62-v7+ (armv7l)
Fri Aug 24 21:34:07 EDT 2018
make -j 4 ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.14.62-v7+/build M=/var/l$
make[1]: Entering directory '/usr/src/linux-headers-4.14.62-v7+'
Makefile:638: arch/armv7l/
Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.14.62-v7+'
Makefile:1918: recipe for target 'modules' failed
make: *** [modules] Error 2
```

kimocoder, 2018-08-25:
You should be using the v5.2.20 branch instead. Report back.

ThijsWithaar, 2018-09-03:
When looking at the header includes:
```
ls /usr/src/linux-headers-4.14.62-v7+/arch/
```
The relevant ARCH is `arm` rather than the `armv7`you've used. The copies of rtl8812au on github which I have compiled successfully all used `arm` as well. Using this might bring you to issue #185.

ThijsWithaar, 2018-09-03:
There seems to be an `#` at line 165 of Makefile which seems wrong. Seems related to the remove line in commit c9e6c422cd34

kimocoder, 2018-09-03:
Yes, will revert the commit this evening

kimocoder, 2018-09-03:
@exotic-fannypack just pushed some changes to the repo, could you try updating with "git pull" and try building it again and confirm the status?

ThijsWithaar, 2018-09-03:
It seems https://github.com/aircrack-ng/rtl8812au/blob/72ceea3533ebfdca56bc5476f97692b6dc8a4fd5/Makefile#L571 is still missing an `ifeq ($(CONFIG_RTL8814A), y) `, which got removed in c9e6c42. This will cause a make error on a trailing endif.

kimocoder, 2018-09-04:
This issue should be resolved, if not.. open it again. Thanks

closed 2018-09-04 by kimocoder

NateZimmer, 2018-11-05:
I've tried 5.15 , v5.2.20 , and /5.3.4. I get the same thing: 
```
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.3.4/source ->
                 /usr/src/rtl8812au-5.3.4

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.14.71-v7+ KSRC=/lib/modules/4.14.71-v7+/build............(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.14.71-v7+ (armv7l)
Consult /var/lib/dkms/rtl8812au/5.3.4/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.14.71-v7+ KSRC=/lib/modules/4.14.71-v7+/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.14.71-v7+ (armv7l)
Consult /var/lib/dkms/rtl8812au/5.3.4/build/make.log for more information.
Finished running dkms install steps.
```

NateZimmer, 2018-11-06:
Using a raspi 3 as well on 4.14.71-v7
ARCH command in the terminal returns armv71 but I think make is suppose to be called with an ARCH of arm. Trying to figure this out.