aircrack-ng/rtl8812au #184: not connecting on 2.4Ghz and unsupported channel on 5Ghz

limones1319, 2018-08-26:
Hi,

I came across two problems. When using 2.4 Ghz it's trying 4 times to do the handshake and then it gives up. When using 5Ghz the SSID does not even show up in the list and it says:

IEEE 802.11 Configured channel (36) not found from the channel list of current mode (2) IEEE 802.11a

Any idea how can I solve this problem?

Thanks,
Gabriel

kimocoder, 2018-08-26:
Need more information here.. which branch, which adapter/chipset?

limones1319, 2018-08-26:
v.5.2.20
TPLINK Archer T4UH

Bus 006 Device 002: ID 2357:010e

limones1319, 2018-08-26:
[ 6822.450061] usb 5-1: new high-speed USB device number 2 using xhci_hcd
[ 6822.578874] usb 5-1: New USB device found, idVendor=2357, idProduct=010e, bcdDevice= 0.00
[ 6822.578885] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6822.578891] usb 5-1: Product: 802.11n NIC
[ 6822.578896] usb 5-1: Manufacturer: Realtek
[ 6822.578901] usb 5-1: SerialNumber: 123456
[ 6830.205746] usb 6-1: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[ 6830.208762] usb 5-1: USB disconnect, device number 2
[ 6830.220235] usb 6-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[ 6830.222218] usb 6-1: New USB device found, idVendor=2357, idProduct=010e, bcdDevice= 0.00
[ 6830.222227] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6830.222232] usb 6-1: Product: 802.11n NIC
[ 6830.222237] usb 6-1: Manufacturer: Realtek
[ 6830.222242] usb 6-1: SerialNumber: 123456

kimocoder, 2018-08-26:
That last output looks good. That chipset is running om 8812au chipset.
Will try re-producing the problem.

limones1319, 2018-08-26:
I forgot to say that I am using hostapd with CentOS Linux release 7.5.1804 (Core) and the .conf that you mentioned on issue #130

fariouche, 2018-10-14:
if possible, can you check with v5.3.4?

fariouche, 2019-01-26:
I do remember having tested that configuration, and 80MHz was not possible in my case because of too many wifi around me.

Channel 36 is pretty crowded, and the driver may not be very good at selecting higher rate if there is noise.
I haven't test this dongle, but with 3 other models, I was able to have a fast speed with my phone.

I do not remember exactly the setup, but you may need to specify the vht_capab variable
vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-123]
(to adapt depending on the reported capabilities of your dongle using iw)

kimocoder, 2019-11-14:
@limones1319 

There are many newer releases available. Move over to v5.6.4.2 and many issues should be resolved, if not.. please re-open this issue or a new one if you stumble upon something. Thanks!

closed 2019-11-14 by kimocoder