aircrack-ng/rtl8812au #185: Error when building v5.3.4/v5.2.20 (armv7l)

MOPO3OB, 2018-08-26:
Good day. v5.1.5 compiles just fine, but using same command for compiling v5.2.20 and v5.3.4 returns an error:

root@kali:~/rtl8812au-v5.3.4# make ARCH=arm
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.14.55+/build M=/root/rtl8812au-v5.3.4  modules
make[1]: Entering directory '/usr/src/kernel-4.14'
  CC [M]  /root/rtl8812au-v5.3.4/core/rtw_cmd.o
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
make[2]: *** [scripts/Makefile.build:329: /root/rtl8812au-v5.3.4/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1522: _module_/root/rtl8812au-v5.3.4] Error 2
make[1]: Leaving directory '/usr/src/kernel-4.14'
make: *** [Makefile:2047: modules] Error 2

Thanks for your hard work. Hopefully, this helps.

kimocoder, 2018-08-26:
@MOPO3OB

Don't use the v5.3.4 yet, it needs some work before it's done.

I pushed an update to v5.2.20 branch, try that Ine again and report back please

kimocoder, 2018-08-26:
I successfully cross-compiled it for ARM after the commit.
Attached build log.

[build.log](https://github.com/aircrack-ng/rtl8812au/files/2321700/build.log)

MOPO3OB, 2018-08-26:
@kimocoder No changes for me.

```
root@kali:~/rtl8812au-v5.2.20# git pull
Already up to date.
root@kali:~/rtl8812au-v5.2.20# git branch -vv
* v5.2.20 00aac0b [origin/v5.2.20] Merge pull request #186 from kimocoder/v5.2.20
root@kali:~/rtl8812au-v5.2.20# make ARCH=arm
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.14.55+/build M=/root/rtl8812au-v5.2.20  modules
make[1]: Entering directory '/usr/src/kernel-4.14'
  CC [M]  /root/rtl8812au-v5.2.20/core/rtw_cmd.o
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
make[2]: *** [scripts/Makefile.build:329: /root/rtl8812au-v5.2.20/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1522: _module_/root/rtl8812au-v5.2.20] Error 2
make[1]: Leaving directory '/usr/src/kernel-4.14'
make: *** [Makefile:1928: modules] Error 2
```

kimocoder, 2018-08-26:
Well, then there must be something with your GCC? I successfully built it with "make ARCH=arm" with no hazzle, on Kali (which you are using too it seems) running the latest of all apt upgrades.

MOPO3OB, 2018-08-26:
Got it. Never had any issues with GCC... Also v5.1.5 builds just fine. Got everything upgraded as well. Will try to investigate this.

kimocoder, 2018-08-26:
Can't really seem to find the issue over here, as you see from my log above.. the build was successfull on ARM here :/ let me know what you find and I'll keep looking too

MOPO3OB, 2018-08-26:
Yeah, I see. Will do. Thank you.

ThijsWithaar, 2018-09-03:
Mmm, same problem here, when building from the rpi3 itself, both soft and hard float seem to be selected.
other archives of rtl8812au don't have that problem.
I diffed the makefiles, but also don't see the cause

kimocoder, 2018-09-03:
Pushed the "endif" fix. "git pull" to update.. report back if the problem is solved

ThijsWithaar, 2018-09-03:
There's also a `ifeq ($(CONFIG_RTL8814A), y)` that got deleted, but I think shouldn't.
After a git pull I got a warning about a trailing endif.

kimocoder, 2018-09-03:
Changes has been committed to both v5.2.20 and v5.3.4 repo.
Use "git pull" and re-build/install the updated driver.

@MOPO3OB could you confirm that armv7l building works now?

ThijsWithaar, 2018-09-03:
It seems https://github.com/aircrack-ng/rtl8812au/blob/72ceea3533ebfdca56bc5476f97692b6dc8a4fd5/Makefile#L571 is still missing an `ifeq ($(CONFIG_RTL8814A), y) `, which got removed in c9e6c42. This will cause a make error on a trailing endif.

MOPO3OB, 2018-09-04:
@kimocoder 5.3.4 builds, 5.2.20 gives "Extraneous endif on line 621".

kimocoder, 2018-09-04:
Ok. Should be fixed now, try again

MOPO3OB, 2018-09-04:
Works now. Thanks.

kimocoder, 2018-09-04:
Thanks to you for reporting! Problem solved, issue closed :+1

closed 2018-09-04 by kimocoder

mrcuongnv, 2018-11-12:
@kimocoder I checked out v5.3.4 and got the same problem.

```
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.14.71-v7+/build M=/home/pi/Downloads/rtl8812au-5.3.4  modules
make[1]: Entering directory '/usr/src/linux-headers-4.14.71-v7+'
  CC [M]  /home/pi/Downloads/rtl8812au-5.3.4/core/rtw_cmd.o
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
scripts/Makefile.build:328: recipe for target '/home/pi/Downloads/rtl8812au-5.3.4/core/rtw_cmd.o' failed
make[2]: *** [/home/pi/Downloads/rtl8812au-5.3.4/core/rtw_cmd.o] Error 1
Makefile:1527: recipe for target '_module_/home/pi/Downloads/rtl8812au-5.3.4' failed
make[1]: *** [_module_/home/pi/Downloads/rtl8812au-5.3.4] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.14.71-v7+'
Makefile:1625: recipe for target 'modules' failed
make: *** [modules] Error 2
```

My kernel is at `4.14.71-v7+`.

mrcuongnv, 2018-11-12:
> I have it compiling on a raspberry pi 3 with a few changes:
> [v5.3.4...ThijsWithaar:v5.3.4](https://github.com/aircrack-ng/rtl8812au/compare/v5.3.4...ThijsWithaar:v5.3.4)
> 
> `CONFIG_PLATFORM_ARM_RPI` should be set to `y`.
> Didn't get around yet to clean them up and make a proper pull request.

Nothing change.

mrcuongnv, 2018-11-12:
@ThijsWithaar nothing change.

mrcuongnv, 2018-11-13:
> I have it compiling on a raspberry pi 3 with a few changes:
> [v5.3.4...ThijsWithaar:v5.3.4](https://github.com/aircrack-ng/rtl8812au/compare/v5.3.4...ThijsWithaar:v5.3.4)
> 
> `CONFIG_PLATFORM_ARM_RPI` should be set to `y`.
> Didn't get around yet to clean them up and make a proper pull request.

Set `CONFIG_PLATFORM_I386_PC = n` as well and it works!