aircrack-ng/rtl8812au #191: After a while it turns off

Log1c1, 2018-08-30:
On the latest version of the driver, the wifi adapter stops working until you reconnect it. I know for sure that it's not an adapter.

renno-bih, 2018-10-26:
Same problem here.
It just stops working randomly. Sometimes can work over 12 hours, sometimes just 5-10 minutes.
It is not that device gets disconnected from USB, it just becomes so slow that nothing cannot work at all.
At this point I have to manually disable WiFi and turn it on again. And it stars working again.

Using AWUS036ACH, OS: Ubuntu 16.04.5 LTS

renno-bih, 2018-10-26:
Hi @francoism90 ,
`dmesg` and `jorunalctl` are not showing anything related to rtl8812 and 5.3.4.

When this happens I can sometimes `ping www.google.com and get response. But I get randomly response from 2-30 seconds, sometimes don't get anything.
After turning off and on wifi everything works again.

Btw, I'm using AWUS036ACH connected to Intel UP Board.

francoism90, 2018-10-26:
@renno-bih Are you running `systemd-resolved`? What happens if you ping to `8.8.8.8`?

renno-bih, 2018-10-29:
Hi @francoism90,
it has nothing to do with DNS. Same issue is present when accessing devices using IP address in local network.

Looks like this problem is **related to 5.8Ghz band**. When this issue happened I tried to manually set my Synology rt1900ac router to 2.4Ghz band only and network started working again. (didn't have to disconnect/connect wifi, it started working automatically) At 2.4Ghz speed is not impressive and there are lot of packets lost. Every second I can notice large number of retries when measuring speed using iPerf3.

```
$ iperf3 -c 192.168.2.31
Connecting to host 192.168.2.31, port 5201
[  4] local 192.168.2.103 port 41620 connected to 192.168.2.31 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec   701 KBytes  5.75 Mbits/sec    1   91.2 KBytes       
[  4]   1.00-2.00   sec   740 KBytes  6.06 Mbits/sec   46   38.5 KBytes       
[  4]   2.00-3.00   sec  1.61 MBytes  13.5 Mbits/sec    0   62.7 KBytes       
[  4]   3.00-4.00   sec  2.24 MBytes  18.8 Mbits/sec    0   84.1 KBytes       
[  4]   4.00-5.00   sec  1.40 MBytes  11.8 Mbits/sec   49   51.3 KBytes       
[  4]   5.00-6.00   sec  1.26 MBytes  10.6 Mbits/sec    0   67.0 KBytes       
[  4]   6.00-7.00   sec  1.23 MBytes  10.3 Mbits/sec    0   81.3 KBytes       
[  4]   7.00-8.00   sec  1.05 MBytes  8.77 Mbits/sec    0   89.8 KBytes       
[  4]   8.00-9.00   sec   489 KBytes  4.01 Mbits/sec    0   92.7 KBytes       
[  4]   9.00-10.00  sec   816 KBytes  6.68 Mbits/sec   10   71.3 KBytes 
```
But looks like it does not freeze permanently on 2.4Ghz.
Hmmm, any idea why it freezes at 5.8Ghz band?

francoism90, 2018-10-29:
What is the distance between the device and the router?
Did you upgrade the firmware? Seems a lot of fixes have been done to fix Wifi:
https://www.synology.com/en-us/releaseNote/RT1900ac

Do you have another AP to test, like your phone?

I had the same issues with a D-Link 868L, after replacing the device with an ASUS AC-RT68U all issues were gone.

renno-bih, 2018-10-29:
Distance is about 50cm at the moment. Could small distance also cause issues?
I've just upgraded my router to the latest version. I'll test stability now.

I can also try using my phone to check if my router is causing problem.

renno-bih, 2018-11-02:
After upgrading rt1900ac router to the newest version of firmware and also moving router further from WiFi adapter it is now working pretty stable :)

renno-bih, 2018-11-06:
UPDATE:
I had the same issue few minutes again. Had to disable 5.8Ghz band on Synology router and enable it again in order to make it work again. 
But it doesn't happen very often.

kimocoder, 2018-12-23:
I think I have found the issue, or actually 2-3 issues related to resume/cycle/idle.
Working on a major rollout with lots of fixes. The list and PR may be seen here https://github.com/aircrack-ng/rtl8812au/pull/246 and I keep crossing 'em off as I go

kimocoder, 2019-02-15:
This issue should be resolved for branch v5.2.20. Open a new issue if it persists

closed 2019-02-15 by kimocoder