aircrack-ng/rtl8812au #200: DKMS status error

brandontict, 2018-09-07:
Alfa AWUS036ACH adapter
Running Kali 2018.3 rolling . 

latest rtl8812au drivers , branch v5.3.4.  

I do believe my main issue is that I've installed several drivers prior to this aircrack's repo and this branch version.
 I do not believe I had fully uninstalled the previous drivers . Despite following readme , using **dkms-uninstall.sh**  and **make uninstall**

For installation I followed readme and Compiled by using **make && make install**, then used the **dkms-install.sh** script. ( Not sure if I was suppose to do both ) 
  
**DKMS status output** 

```
dkms status
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/rtl8812au/5.2.20/source/dkms.conf does not exist.
```
Under /usr/src/ I do have a  **rtl8812au-5.2.20.2** folder as well as 5.2.20 

**lsmod output**  

```
 lsmod | grep 88
88XXau               2347008  0
bluetooth             626688  5 btrtl,btintel,btbcm,btusb
8812au               1622016  0
cfg80211              774144  5 iwldvm,8812au,iwlwifi,mac80211,88XXau
usbcore               290816  8 xhci_hcd,ehci_pci,usbhid,ehci_hcd,8812au,btusb,xhci_pci,88XXau
```
Looks like I have 88XXau and 8812au loaded. 

At the moment card is working. Although injection is laggy . Also airmon-ng is laggy . 
I want to make sure I am using latest branch. Any suggestions ? 

Appreciate all the work done on aircrack suite as well as releasing the rtl* drivers with injection.

kimocoder, 2018-09-07:
Do a "updated" and "locate 8812au.ko" delete the files found by locate.
Also, do a "modprobe -r 8812au"

brandontict, 2018-09-07:
Thanks.  Followed your direction . 

**lsmod output** 
```
lsmod | grep -i 88
bluetooth             626688  5 btrtl,btintel,btbcm,btusb
88XXau               2347008  0
cfg80211              774144  4 iwldvm,iwlwifi,mac80211,88XXau
usbcore               290816  7 xhci_hcd,ehci_pci,usbhid,ehci_hcd,btusb,xhci_pci,88XXau
```
8812au is now removed. 

airmon-ng & aireplay-ng are both much faster. Injection working 3x better as well.  

One more issue that I wasnt sure if I needed start a new thread for. 
 I am unable to adjust tx-power 

```
ip link set wlan1 down
iwconfig wlan1 txpower 30
ip link set wlan1 up 
Error for wireless request "Set Tx Power" (8B26) :
    SET failed on device wlan1 ; Operation not permitted.
``` 

I did not find regdb inside the repo folder, I know there are plenty of articles out there on how to change tx-power. Just wondering If I missed something simple.
Any insight is appreciated. 

Thanks again for the help on multiple modules .

kimocoder, 2018-09-08:
v5.3.4 is a "work in project" driver. everything isn't really ready. most is, some not 
Great you got ut working, closing down.

closed 2018-09-08 by kimocoder