aircrack-ng/rtl8812au #201: Installation problem after upgrading Linux kernel (8814au driver)

Alamoz, 2018-09-11:
After upgrading the Linux kernel the driver no longer works. I've had this happen several times now. I had a fix for the problem which I now can't find, should have submitted a patch to your install script when I had it.

The problem seems to be with a magic number. The magic number from the previous kernel seems to have gotten inserted into the build. I completely removed all the older kernels, including the old build/header directories, did a 'make clean', ran the uninstall and then the install script.

Output of dmesg for the upgraded driver:

[ 2067.899782] 8814au: version magic '4.15.0-33-generic SMP mod_unload ' should be '4.15.0-34-generic SMP mod_unload '

Alamoz, 2018-09-11:
So the problem was there was a 8814au.ko driver in /lib/modules/4.15.0-34-generic/updates/ containing the wrong mod_unload command. I removed /lib/modules/4.15.0-34-generic/updates/8814au.ko, did a 'make clean', ran 'sudo ./dkms-remove.sh' then ran sudo ./dkms-install.sh.

There was also a 8814au.ko driver in /lib/modules/4.15.0-34-generic/kernel/net/wireless/ perhaps a result of building the driver using 'make RTL8814=1' and 'sudo make install RTL8814=1' rather than using the installation script. Strangely, there is now also a 8812au.ko driver in /lib/modules/4.15.0-34-generic/updates/

closed 2018-09-11 by Alamoz

kimocoder, 2018-09-11:
So which branch was this? v5.1.5 ?