aircrack-ng/rtl8812au #208: Something for the virtual interface?

LanzoT, 2018-10-03:
Hi! I have a D-Link DWA-182 vC1 that has the 8812au chip.
I've installed the 5.2.20 version of the driver and it works flawlessy, except for the already mentioned error with hostapd, that is when i try to add a virtual interface from the D-Link and use it in hostapd, it gives me an error that is could not find any kernel drivers for it, and so I can't use it for an EvilTwin attack.

I've been searching in the web for a solution or something like that, and I've found someone who said that the problem was in the concurrent_mode of the driver and in the source files while trying to building it. Here's the links:

https://randomstuffidosometimes.blogspot.com/2016/03/rtl8192cu-and-rtl8188cus-in-station-and.html

I know the chip is different and the article is quite old, but maybe could help you to figure out what the problem is?

kimocoder, 2018-10-03:
The chip is different, but Realtek drivers is almost all the same. Thanks, will read when I got time.

LanzoT, 2018-10-03:
I found another useful link, I hope, to handle the hostpad problem.
I found an old fork of Fluxion for drivers that not support virtualization (it supports rtl8814au) and it kinda works. I tried it, I can start a fake ap on the same device that I used for monitoring, but it seems that i cannot access the fake login page, maybe it's just me tho.

Here's the link, maybe it'd be useful for you:

https://github.com/nicshaca/Fluxion-awus1900

LanzoT, 2018-10-09:
Yeah, my intention is to have AP and Station at the same time in the same interface and that'd be possible (as I understood) only using a virtual interface (as  stated here: https://github.com/FluxionNetwork/fluxion/wiki/Starting-access-point-hangs! in the "Verify Virtual Interface Functionality" part).

All I'm trying to do now is finding useful sources online to allow devs to enable virtual interface even for this rtl driver, as hostpad always says it can't find any kernel driver fro the virtual interface.

Sorry for my poor english!

kimocoder, 2018-10-09:
"virtual Interface" has the ability to create different interfaces. I know about fluxion, what wifite and aircrack-ng did is they added support for the drivers (not using airmon-ng) but instead setting it manually.

virtual Interface support is a prority cause then the drivers could support more tools without the hazzle. you may/should consult fluxion devs about ut, ref. to this post, until er got ut fixed which could taket some time, as I've seen the code is in place, but not figured out the issue with it.

LanzoT, 2018-10-09:
Could the problem be just with hostapd? 'Cause I've seen a modified driver for it for a specific rtl chip (https://github.com/pritambaral/hostapd-rtl871xdrv).

Or maybe I'm just wrong lol, in case sorry.

kimocoder, 2018-10-10:
This report should be linked to #93

kimocoder, 2019-02-15:
We're having this as high priority still, but we're moving a notch up to a newer base anytime so I'll close this as we got a own thread (issue) for it which is well documented.

closed 2019-02-15 by kimocoder

visbs, 2019-08-23:
@Le0T  were u successful doing the same , i.e where u be able to use the concurrent mode in this driver?? If yes do let me know. 
Thank You

fariouche, 2019-08-30:
just for information, I just was able to register 3 wlan virtual interfaces successfully.
My use case is dual band AP. I can see both of them using hostapd (5GHz and 2.4GHz).
Still unable to do the association, but this is most probably a hostapd configuration issue.
I'm checking and let you know

fariouche, 2019-08-30:
by the way, I've done this on the 5.6.4.1 branch

Medullitus, 2022-02-02:
Help please. I have the same chipset please help!!!