aircrack-ng/rtl8812au #210: Can't Install On Odroid C2 Running Kali

jsl303, 2018-10-05:
I have AWUS036ACH (RTL8812AU), and I'm trying to get it going on odroid c2 running kali 2018.3.
I followed the instruction from this issue on kali-arm-build-scripts.
https://github.com/offensive-security/kali-arm-build-scripts/issues/117
make mrproper
cp /usr/src/odroid-c2.config /usr/src/kernel/.config
make prepare
make modules_prepare  
Then I cloned the rtl8812au repo, and run "sudo ./dkms-install.sh".
However, I got "Error! Bad return status for module build on kernel: 3.14.79 (aarch64)".
Here's the log.
https://paste.ee/d/7hk1w
What might be causing this?
Thanks for your help!

steev, 2018-10-05:
My guess is (without looking closely at the code) - https://github.com/aircrack-ng/rtl8812au/commit/8c6132a401e25be88a37198710b3b6c6b8f00117 doesn't take into account arm64.

kimocoder, 2018-10-05:
Feel free to make a working commit. I don't have snu aarch64/arm64 devices for testing

steev, 2018-10-12:
I've been looking into this a bit.  Their makefile is absolutely horrendous.  I will keep poking at it but this is an extremely low priority for me right now.

jsl303, 2018-10-13:
@steev, thank you for your willingness to help! This would be awesome if we could get this work with Odroid running Kali!

kimocoder, 2018-12-23:
Pushing a collection lf both major and minor fixes in a week, should make the Odroid C2 and RPI3 issues go away. Follow it https://github.com/aircrack-ng/rtl8812au/pull/246

kimocoder, 2019-01-10:
Disabled the "CONFIG_MP_VHT_HW_TX_MODE" which should fix compilation for those chipsets.
Update the driver and build/install/restart and please report back if that fixes the issue.

p-stepn, 2019-01-26:
I'm on an Odroid C2 as well, tried both Ubuntu (3.16.62-35) and Kali (3.14.79) with branch 5.2.20.
Looing at the dkms log, both throw the same set of errors, logged here: https://paste.ee/p/tmSlz
In short, compilation fails on ioctl_cfg80211.o, function cfg80211_rtw_get_station.
Any suggestions on what to try? Many thanks!

[image]

kimocoder, 2019-01-26:
Aaah, it's the get_station function that's not working with Odroid C2.
I'll go add a switch to it :+1:

kimocoder, 2019-02-15:
This issue should be resolved, if not.. open it again please :+1: 
Lots of RPI/ODROID and generic ARM/ARM64 support has been added and the new I/O controllers that caused problems with ARM devices is disabled. Thanks

closed 2019-02-15 by kimocoder