aircrack-ng/rtl8812au #217: v5.3.4 build and v5.1.5  not working on Ubuntu 18.04 on 4.19.0-041900rc7-generic

NikolaiT, 2018-10-09:
Hi Everyone,

I have a wlan adapter AWUS036ACH which I want to use in monitor mode to inject frames. 

```
$ uname -r
4.19.0-041900rc7-generic
```

I tried installing the branch v5.3.4 and v5.1.5. Installation is successful and the driver is loaded and I can use the driver in managed mode:

```
$ dkms status
rtl8812au, 5.1.5: added
```

```
$ iwconfig
wlx00c0caa51413  IEEE 802.11bgn  ESSID:"TC-3C110"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.462 GHz  Access Point: XX:XX:XX:XX:XX:XX
          Bit Rate:144.4 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=96/100  Signal level=54/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

If I try to set the interface in monitor mode, I get:

```
$ sudo airmon-ng check kill

Killing these processes:

  PID Name
 5229 wpa_supplicant
 6636 avahi-daemon
 6647 avahi-daemon

$ sudo ip link set wlx00c0caa51413 down
$ sudo iw dev wlx00c0caa51413 set type monitor
command failed: Operation not supported (-95)
```

Why does it fail to put my device in monitor mode? The error message is the same with branch v5.3.4 and v5.1.5. 

When I used a mainline kernel  4.18.* from august 2018, the driver worked perfectly. Could it be that the driver is not compatible with kernel  4.19.* ?

Any ideas how to investigate the problem?

kimocoder, 2018-10-10:
Try:

ifconfig <dev> down
iwconfig <dev> mode monitor

darkdragon-001, 2018-10-23:
See if #226 fixes it.

kimocoder, 2019-02-15:
v5.1.5 is deprecated here, we have focus on the newer releases.
v5.3.4 is back in active development again too, and it got backported API support for older kernels and support above kernel v5.0. If you want to use only the v5.1.5, then it's easy to just squash the patches cherry-picked. Thanks

closed 2019-02-15 by kimocoder