aircrack-ng/rtl8812au #219: AWUS036ACH slow speed issues

renno-bih, 2018-10-12:
Hi,
I tried both versions, v5.2.20 and v5.3.4 but cannot manage to get high speed with AWUS036ACH (RTL8812AU).
By using iperf I measured speeds of around **60-90Mbits/sec.**
I'm sure that router (Synology RT2600ac) is not bottleneck because my internal laptop WiFi 802.11AC card can achieve about 400Mbits/sec.

> burger@burger-UP-CHT01:~$ iperf3 -c 192.168.2.31
> Connecting to host 192.168.2.31, port 5201
> [  4] local 192.168.2.103 port 48042 connected to 192.168.2.31 port 5201
> [ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
> [  4]   0.00-1.00   sec  9.50 MBytes  79.7 Mbits/sec    0    211 KBytes
> [  4]   1.00-2.00   sec  8.84 MBytes  74.1 Mbits/sec   10    160 KBytes
> [  4]   2.00-3.00   sec  8.05 MBytes  67.6 Mbits/sec    0    195 KBytes
> [  4]   3.00-4.00   sec  9.20 MBytes  77.1 Mbits/sec    1    177 KBytes
> [  4]   4.00-5.00   sec  8.98 MBytes  75.3 Mbits/sec    0    208 KBytes
> [  4]   5.00-6.00   sec  8.36 MBytes  70.1 Mbits/sec    0    208 KBytes
> [  4]   6.00-7.00   sec  8.75 MBytes  73.4 Mbits/sec    0    208 KBytes
> [  4]   7.00-8.00   sec  9.32 MBytes  78.2 Mbits/sec    0    208 KBytes
> [  4]   8.00-9.00   sec  9.84 MBytes  82.5 Mbits/sec    0    208 KBytes
> [  4]   9.00-10.00  sec  9.32 MBytes  78.2 Mbits/sec    0    208 KBytes
> 
> [ ID] Interval           Transfer     Bandwidth       Retr
> [  4]   0.00-10.00  sec  90.2 MBytes  75.6 Mbits/sec   11             sender
> [  4]   0.00-10.00  sec  90.0 MBytes  75.5 Mbits/sec                  receiver
> 

Also, no matter if I'm connecting WiFi card using USB2.0 or USB3.0 connector I'm getting similar speeds. (by checking USB ports it says that device operates at 5000M)

`iwconfig `gives following results: 

> wlx00c0caa51893  IEEE 802.11abgn  ESSID:"NSoftR&D"
>           Mode:Managed  Frequency:5.5 GHz  Access Point: 00:11:32:72:1B:71
>           Bit Rate=**867 Mb/s**   Tx-Power=18 dBm
>           Retry short limit:7   RTS thr:off   Fragment thr:off
>           Power Management:off
>           Link Quality=70/70  Signal level=-11 dBm
>           Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
>           Tx excessive retries:0  Invalid misc:0   Missed beacon:0

I tried this USB card on Windows and I can get speeds of about 150-200Mbits/sec.
I'm using Ubuntu 16.04, having AWUS036ACH connected to USB3.0 on Intel Up Board.

Any ideas what could be wrong?

fariouche, 2018-10-14:
Not sure about the issue, but at least with v5.3.4 I can achieve 370Mbits with the exact same dongle.
I is possible that the kernel version may affect the driver's behavior, since they changed many things, especially compilation flags. So maybe a bad flag in the driver somewhere is possible.

Other possibility: hardware interferences. Some usb3 HW may interfere with wifi frequencies, and the driver is then dropping the speed to keep the link

kimocoder, 2019-06-17:
@renno-bih Now, 8812au users should move over to the more stable v5.6.4 branch which got phydm updates and more. Overall a better choice. :+1:

kimocoder, 2019-06-17:
Open the issue again if the problem persists.

closed 2019-06-17 by kimocoder