aircrack-ng/rtl8812au #222: Can't set monitor in odroid c1+ (kernel 3.10.107)

youmych, 2018-10-16:
I've compile the driver for hardkernel odroid c1+ (kali rolling) with kernel 3.10.107. I will use it with D-Link DWA-182 and Linksys WUSB6300 devices. 
Driver was loaded successfully and i see wlanX adapters. 

But I couldn't setup devices in monitor mode:
```
5: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 10:62:eb:44:27:ab brd ff:ff:ff:ff:ff:ff
```
```
root@kali:~# ip link set wlan0 down
root@kali:~# iw wlan0 set monitor control
command failed: Operation not supported (-95)
```
All devices work properly if the driver compiled on x86_64 kali linux and linux mint 19.

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)

youmych, 2021-05-12:
> pls consider closing the issue, when it's solved by now :)

unfortunately, I can't check the issue, it's not actually for me.

closed 2021-05-12 by youmych