aircrack-ng/rtl8812au #223: AP mode supported ?

Bob36180, 2018-10-20:
Hello,
I just wanted to know if there were a driver release wich supports AP mode AND monitor mode ? Because I try to set up a Rogue AP with the AWUS036ACH but I can't because there isn't AP mode...
Thanks for advance !!

kimocoder, 2019-02-15:
@Bob36180 which branch are you using? They support AP mode, monitor mode, frame injection, MESH mode, AP-STA.. but combinations doesn't it support.

chises, 2019-02-27:
hello,
my question to clarify:
we are not abel to use monitor mode an AP mode in parallel with that driver?

kcdtv, 2019-03-04:
I think that Kimocoder answered you: "Combination" (various mode "in parallel") is not supported. 
The same happens if you want to use your device as wifi repeater ( AP and client at the same time): It won't work. 
I guess that the first step to get there would be to be able to get working virtual interfaces which is very tricky. 
You can use airebase-ng to do soft AP with monitor mode, it works.

closed 2019-08-17 by kimocoder