aircrack-ng/rtl8812au #225: rtl8814au always reporting 300 Mb/s tx bitrate

seeul8er, 2018-10-22:
My AWUS1900 is always reporting a bitrate of 300 Mbit/s

OS: Ubuntu 18.04
Tested drivers: 5.1.5, 5.2.20, 5.3.4
Freq: 5180
Mode: Managed/Connected to AP, two walls in between

bitrate as seen when running: `iwconfig` or `iw dev xxx link`
It does not change when I remove the antennas so I guess it is sort of hardcoded in the driver.

dernyn, 2018-10-23:
i have the same problem, but it's 450Mb/s for me.

francoism90, 2018-10-23:
Just to be sure: is this on 802.11ac 5GHz or 2.4GHz?

seeul8er, 2018-10-23:
@francoism90 I am connected to a 802.11n 5GHz access point. Max possible data rate is 300Mbit/s. So in theory the reported value is possible, but it seems the driver just reports the max data rate rather than the actual one (stays the same when I detach the antennas).

francoism90, 2018-10-27:
@seeul8er I don't know (same value reporting here on 802.11n), better ask to the developers. :)

fariouche, 2018-11-28:
I know that I can go up to the max in my case, but I'm working essentially in AP mode.
The speed selection is very peeky, and if there is any interference, it will switch down.

Also, it depends on the AP capabilities, so be sure to check the AP's spec. Depending on the number of spatial streams it can handle. I believe that your AP is only 2 SS (2 antennas for 5GHz) and 300Mbps is for ShortGI, 40MHz, MCS7 (pretty standard) and can go up to 400Mbps. (MCS 9)
Using a higher MCS is subject to noise, and I find the driver to prefer MCS7 easily over MCS9 :(

ValZapod, 2019-01-26:
> 802.11n 5GHz

Ooh. My. If it is really the case, it is very rare configuration...

QianrenLi, 2023-10-29:
> pls consider closing the issue, when it's solved by now :)

HI @cedricbambooza, I wonder is this problem is solved or not since I also can not get changing MCS value even when the iperf shows tx rate changing. Please tell me any method to get exact MCS. Thank you very much.