aircrack-ng/rtl8812au #227: Injection mode not working v5.1.5 rtl8814au And add support for AP+Jammer mode

souravdas142, 2018-10-27:
**On version 5.1.5 Injection mode not working with airoplay
 Please add support AP+Jammer mode on Realtek rtl8814au chipset for fluxion .**

 Can you please fix the issue to put the card(rtl8814au) to monitor mode in traditional way (airmon-ng start WlanInterfaceName) instead of iwconfig ? or this is card issue not the driver?

kimocoder, 2018-10-27:
Newest aircrack-ng suite for support for this driver, so update ut to get support

souravdas142, 2018-10-27:
> Newest aircrack-ng suite for support for this driver, so update ut to get support

Thanks for quick reply. which driver to install 5.2.20 or 5.3.4 which is more stable?  **Can you Please add support AP+Jammer mode on Realtek rtl8814au chipset for fluxion .**

kimocoder, 2018-10-27:
v5.2.20 should be used atm. v5.3.4 isn't finished yet

souravdas142, 2018-10-31:
Thanks Everything is working fine but one issue detected airmon-ng has trouble to put monitor mode with very long interface name. When airmon-ng goes to rename interface name from wlx00c0caa50e34 to wlanXmon system hang occurred. Everything freezes until a reboot. I don't know if it is a aircrack-ng suite problem or driver. If it is a driver problem please fix it. (Testing on Ubuntu 18.04 LTS on latest aircrack-ng 1.4)

closed 2019-02-15 by kimocoder