aircrack-ng/rtl8812au #228: AWUS036ACH Problem in Kali Linux 2018.3 (make errors)

JeroenVBH, 2018-10-28:
``root@kali:~/rtl8812au# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.18.0-kali2-amd64/build M=/root/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.18.0-kali2-amd64'
  CC [M]  /root/rtl8812au/core/rtw_cmd.o
In file included from /root/rtl8812au/include/osdep_service.h:41:0,
                 from /root/rtl8812au/include/drv_types.h:32,
                 from /root/rtl8812au/core/rtw_cmd.c:22:
/root/rtl8812au/include/osdep_service_linux.h: In function ‘_init_timer’:
/root/rtl8812au/include/osdep_service_linux.h:284:8: error: ‘_timer {aka struct timer_list}’ has no member named ‘data’
  ptimer->data = (unsigned long)cntx;
        ^~
/root/rtl8812au/include/osdep_service_linux.h:285:2: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  init_timer(ptimer);
  ^~~~~~~~~~
  _init_timer
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-4.18.0-kali2-common/scripts/Makefile.build:323: /root/rtl8812au/core/rtw_cmd.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.18.0-kali2-common/Makefile:1531: _module_/root/rtl8812au] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.18.0-kali2-amd64'
make: *** [Makefile:1732: modules] Error 2
root@kali:~/rtl8812au#

kimocoder, 2018-10-28:
which branch?

JeroenVBH, 2018-10-29:
> which branch?
Ok I stopped and did a clean install. This morning I updated everything, installed DKMS and Linux Headers and now it does "make File" (branch 5.4.3).

Problem is my AWUS036ACH`s blue LED stays off and IWCONFIG or AIRMON-NG does not show a WLAN0 (or any other). 
I managed to get it working with RTL8812AU from Gordboy version 5.2.20.2 but this version won`t let me use GUI style Kali tools like Airgeddon and Ghost-Phisher.

Your help is much appreciate!

JeroenVBH, 2018-10-29:
root@kali:/usr/src/rtl8812au-5.3.4# make install
install -p -m 644 88XXau.ko  /lib/modules/4.18.0-kali2-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 4.18.0-kali2-amd64
root@kali:/usr/src/rtl8812au-5.3.4#

kimocoder, 2018-11-01:
The first error reported doesn't seem to be the v5.3.4 branch, but rather the master or something cause the init_timer fix was for the kernel v4.15 that was added to most branches besides the "master" which should not be used. But you got it working?

kimocoder, 2019-02-15:
This issue has been resolved. Kernel v4.15 _init_timers is added some time back :+1:

closed 2019-02-15 by kimocoder