aircrack-ng/rtl8812au #229: Question : which branch to use for WifiBroadcast ?

Yes21, 2018-10-30:
Hi, 
We are preparing a new version of [WifiBroadcast](https://github.com/bortek/EZ-WifiBroadcast) with support for AWUS036AC/H and AWUS1900. WBC works on raspberry Pis and uses only usb wifi boards with monitor mode and packet injection. So we want to support these two boards which are dual bands and very promising.
We could already install the 5.2.20 branch on raspian (kernel 4.14.66)
Do you think the 5.3.4 will be better, or is it too early to use it ?

Many thanks for your advice.

kimocoder, 2018-11-01:
Great that you want to add support for these drivers. Unfortunately the v5.3.4 isn't quite ready yet, do the v5.2.20(.2) should be used for now.

Yes21, 2018-11-01:
Thanks for your kind answer.
Finally, the v5.2.20 doesn't compile anymore on latest raspbian version !
So we are testing now the v5.3.4 which compiles well on 4.14.78.

We only use monitor mode ...

Yes21, 2018-11-06:
We are now able to install 5.2.20 on RPI kernel 4.14.xx.
This driver is now part of our latest beta Wifibroadcast version !
Many users have now very happy to be able to use AWUS1900 and AWUS03ACH with next EZ-WBC.
Thanks to you ...

I will now test with 5.3.4.

Yes21, 2018-11-20:
We have now tested v5.3.4 branch. It seems to work for us as good as 5.2.20.
We have always the same [problem](https://github.com/aircrack-ng/rtl8812au/issues/116) with packet loss with 8812 (not with 8814).

So can you help us to make our definitive choice between 5.2.20 and 5.3.4 for our next EZ-WBC release :
- does 5.3.4 has real improvements in monitor mode ?
- does it has still known issues ?

Thanks for this great driver.

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)