aircrack-ng/rtl8812au #230: compiler error with 5.2.20 branch

howudodat, 2018-11-01:
Trying to build on a gateworks newport SBC.  The 5.1.5 branch builds fine

```
make RTL8814=1 -j4
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.14.4-ged9e7d8/build M=/root/rtl8812au  modules
make[1]: Entering directory '/root/linux-newport'
  CC [M]  /root/rtl8812au/core/rtw_cmd.o
  CC [M]  /root/rtl8812au/core/rtw_security.o
**gcc: error: unrecognized command line option '-mhard-float'**
scripts/Makefile.build:314: recipe for target '/root/rtl8812au/core/rtw_cmd.o' failed
make[2]: *** [/root/rtl8812au/core/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
  CC [M]  /root/rtl8812au/core/rtw_debug.o
gcc: error: unrecognized command line option '-mhard-float'
scripts/Makefile.build:314: recipe for target '/root/rtl8812au/core/rtw_security.o' failed
make[2]: *** [/root/rtl8812au/core/rtw_security.o] Error 1
gcc: error: unrecognized command line option '-mhard-float'
scripts/Makefile.build:314: recipe for target '/root/rtl8812au/core/rtw_debug.o' failed
make[2]: *** [/root/rtl8812au/core/rtw_debug.o] Error 1
Makefile:1503: recipe for target '_module_/root/rtl8812au' failed
make[1]: *** [_module_/root/rtl8812au] Error 2
make[1]: Leaving directory '/root/linux-newport'
Makefile:1933: recipe for target 'modules' failed
make: *** [modules] Error 2

uname -a
Linux MC 4.14.4-ged9e7d8 #4 SMP PREEMPT Mon Jul 9 10:37:14 PDT 2018 aarch64 aarch64 aarch64 GNU/Linux

gcc --version
gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
```

ThijsWithaar, 2018-11-17:
Might the be a duplicate of #233?

olance, 2018-12-02:
@ThijsWithaar I'd say it's not: I've tried applying your patch and face the same issue.

Building directly on a Raspberry Pi 3 model B (arm64) with linux 4.14.71-v8.

What made it compile correctly though was to re-comment the line in the Makefile that adds `EXTRA_CFLAGS += -mfloat-abi=softfp`. It is present but commented out in the v5.1.5 branch, and I guess it's the reason it builds successfully.

I'm guessing that as long as you're not cross-compiling, `gcc` will use the floating-point option from your building machine and it should work. (it's still building on my Pi right now so I can't tell yet for sure ^^)

The biggest question I have is: why doesn't `gcc` recognize this option when it's clearly documented as an ARM-only option? Would compiling `gcc` manually with configure option `--with-float` make it recognize the option?

kimocoder, 2018-12-03:
And what was the solution? Please inform us if we need to do some changes a I've don't for the hardware to test with

olance, 2018-12-04:
@kimocoder  As I said earlier:

> What made it compile correctly though was to **re-comment the line in the Makefile that adds EXTRA_CFLAGS += -mfloat-abi=softfp**. It is present but commented out in the v5.1.5 branch, and I guess it's the reason it builds successfully.

The one downside with doing this, is that if you're cross-compiling on a host that uses a different float ABI setting, there's a high chance the built drivers won't be compatible with your target.  
I couldn't understand why in my case and OP's one, GCC doesn't recognize the `float-abi` option when it's clearly documented in GCC's docs 😕   
Maybe GCC needs to be compiled manually with a specific configuration for this to work?

howudodat, 2019-01-04:
commenting out EXTRA_CFLAGS += -mfloat-abi=softfp caused me another compiler error
new error: due to ‘-mgeneral-regs-only` in the default arm64 kernel source Makefile

recompiling the kernel with -mgeneral-regs-only commented out int he kernel makefile and then recompiling the driver succeeded

The card is now registered, can scan, sees 2G and 5G networks, but can only connect to 2G networks

kimocoder, 2019-01-04:
Ok. Thanks for the information!

There's alot going on now, getting ready to push a major PR with lots of fixes/changes.
It just needs some more work before it's done. Again, thanks alot!

kimocoder, 2019-02-15:
This issue has been resolved. Closing :+1:

closed 2019-02-15 by kimocoder