aircrack-ng/rtl8812au #233: Floating point issue on arm64

svpcom, 2018-11-07:
rtl8812au driver has some portions of code with floating point (only used when CONFIG_MP_VHT_HW_TX_MODE is enabled). But on arm64 there are no support for soft floating-point and HW fp is disabled in kernel. Are there any workaround to build driver with enabled CONFIG_MP_VHT_HW_TX_MODE or only rewriting ``core/rtw_mp.c`` to use fixed-point arithmetics?

ThijsWithaar, 2018-11-17:
The 5.2.20 branch has the proper floating-point setting, introduced in https://github.com/aircrack-ng/rtl8812au/commit/1603b05c0f5c5547b5901d908024c308e4c088f5.

With the following changes, the 5.3.4 branch compiles on my raspberry-pi3 (arm32):
https://github.com/aircrack-ng/rtl8812au/compare/v5.3.4...ThijsWithaar:v5.3.4

svpcom, 2018-11-17:
This only works on x86 (where fpu is allowed in kernel using kernel_fpu_begin/kernel_fpu_end) or on arm32 where there are software fpu emulation ( -mfloat-abi=softfp ). But on arm64 (for example nanopi neo2) there are no in-kernel fpu support and software fpu. So the only way to disable CONFIG_MP_VHT_HW_TX_MODE functionality. Does this function has any meaning for packet injection or it is only using in normal mode?

denni-isl, 2018-11-27:
Get similar error message on rk3399 aarch64 rockpro64 (mini computer :-) ) 
-------------
include/linux/sched.h:1211:1: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
 const struct sched_group_energy * const(*sched_domain_energy_f)(int cpu);
 ^~~~~
/home/rock64/bin/driverar/rtl8812au/core/rtw_mp.c: In function ‘rtfloor’:
/home/rock64/bin/driverar/rtl8812au/core/rtw_mp.c:36:12: error: ‘-mgeneral-regs-only’ is incompatible with floating-point argument
 static int rtfloor(float x) {
            ^~~~~~~
scripts/Makefile.build:277: recipe for target '/home/rock64/bin/driverar/rtl8812au/core/rtw_mp.o' failed
make[2]: *** [/home/rock64/bin/driverar/rtl8812au/core/rtw_mp.o] Error 1
Makefile:1474: recipe for target '_module_/home/rock64/bin/driverar/rtl8812au' failed
make[1]: *** [_module_/home/rock64/bin/driverar/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.154-1124-rockchip-ayufan-ged3ce4d15ec1'
Makefile:1918: recipe for target 'modules' failed
make: *** [modules] Error 2
----------------------------
Thanks,

satmandu, 2019-01-01:
Sadly float in kernel modules always seems to fail due to ‘-mgeneral-regs-only` in the default arm64 kernel source Makefile.

Does anyone want to replace this float code in this one file to enable this driver working properly on arm64, or barring that, find a way to get float enabled despite the kernel being compiled to not use float? (All arm64 systems should have float in hardware, I think.)

kimocoder, 2019-01-01:
CONFIG_MP_VHT_HW_TX_MODE is disabled in a pull request coming up

satmandu, 2019-01-01:
FYI 5Ghz DFS channels are not working with arm64 on 4.20, possibly with all versions I am seeing of this driver, including yours @kimocoder .

howudodat, 2019-01-04:
FYI ARMV8 supports hardware floating point.  recompiling the kernel with -mgeneral-regs-only removed from the makefile and compiling rtl8812 driver with  -mfloat-abi=softfp removed seems to work fine

+1 on the 5GHz DFS channels not working

kimocoder, 2019-01-04:
DFS is disabled, check "include/autoconf.h" you may enable it and compile/build/install and reboot (or "modprobe -r 88XXau && modprobe 88XXau")

Try it, test again and report back then we've got it fixed by your help

kimocoder, 2019-01-04:
@Re4son may be some interesting for you here

howudodat, 2019-01-04:
not sure this will help, but here is my syslog:

```Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.7716] device (wlan0): Activation: starting connection 'CARLSON5' (645a25da-d9ba-4160-be4c-4a9f58f96065)
Jan  4 19:02:37 localhost dbus[1784]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service'
Jan  4 19:02:37 localhost NetworkManager[1876]: <warn>  [1546628557.7758] sup-iface[0x16bad8a0,wlan0]: connection disconnected (reason -3)
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.7762] device (wlan0): supplicant interface state: completed -> disconnected
Jan  4 19:02:37 localhost systemd[1]: Starting Network Manager Script Dispatcher Service...
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.7879] device (wlan0): state change: disconnected -> prepare (reason 'none') [30 40 0]
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.7911] device (wlan0): state change: prepare -> config (reason 'none') [40 50 0]
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.7928] device (wlan0): Activation: (wifi) access point 'CARLSON5' has security, but secrets are required.
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.7929] device (wlan0): state change: config -> need-auth (reason 'none') [50 60 0]
Jan  4 19:02:37 localhost dbus[1784]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Jan  4 19:02:37 localhost systemd[1]: Started Network Manager Script Dispatcher Service.
Jan  4 19:02:37 localhost nm-dispatcher: req:1 'down' [wlan0]: new request (1 scripts)
Jan  4 19:02:37 localhost nm-dispatcher: req:1 'down' [wlan0]: start running ordered scripts...
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.8292] device (wlan0): state change: need-auth -> prepare (reason 'none') [60 40 0]
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.8308] device (wlan0): state change: prepare -> config (reason 'none') [40 50 0]
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.8318] device (wlan0): Activation: (wifi) connection 'CARLSON5' has security, and secrets exist.  No new secrets needed.
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.8320] Config: added 'ssid' value 'CARLSON5'
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.8321] Config: added 'scan_ssid' value '1'
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.8322] Config: added 'key_mgmt' value 'WPA-PSK'
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.8322] Config: added 'auth_alg' value 'OPEN'
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.8322] Config: added 'psk' value '<omitted>'
Jan  4 19:02:37 localhost NetworkManager[1876]: <info>  [1546628557.8349] sup-iface[0x16bad8a0,wlan0]: config: set interface ap_scan to 1
Jan  4 19:02:43 localhost wpa_supplicant[2159]: wlan0: Trying to associate with 60:a4:4c:dd:5d:8c (SSID='CARLSON5' freq=5240 MHz)
Jan  4 19:02:43 localhost wpa_supplicant[2159]: wlan0: Association request to the driver failed
Jan  4 19:02:43 localhost NetworkManager[1876]: <info>  [1546628563.1438] device (wlan0): supplicant interface state: disconnected -> scanning
Jan  4 19:02:47 localhost wpa_supplicant[2159]: wlan0: Trying to associate with 60:a4:4c:dd:5d:8c (SSID='CARLSON5' freq=5240 MHz)
Jan  4 19:02:47 localhost wpa_supplicant[2159]: wlan0: Association request to the driver failed
Jan  4 19:02:47 localhost NetworkManager[1876]: <info>  [1546628567.2636] device (wlan0): supplicant interface state: scanning -> disconnected
Jan  4 19:02:47 localhost NetworkManager[1876]: <info>  [1546628567.7440] device (wlan0): supplicant interface state: disconnected -> scanning
Jan  4 19:02:51 localhost wpa_supplicant[2159]: wlan0: Trying to associate with 60:a4:4c:dd:5d:8c (SSID='CARLSON5' freq=5240 MHz)
Jan  4 19:02:51 localhost wpa_supplicant[2159]: wlan0: Association request to the driver failed
Jan  4 19:02:51 localhost NetworkManager[1876]: <info>  [1546628571.8661] device (wlan0): supplicant interface state: scanning -> disconnected
Jan  4 19:02:52 localhost NetworkManager[1876]: <info>  [1546628572.8448] device (wlan0): supplicant interface state: disconnected -> scanning
Jan  4 19:02:56 localhost wpa_supplicant[2159]: wlan0: Trying to associate with 60:a4:4c:dd:5d:8c (SSID='CARLSON5' freq=5240 MHz)
Jan  4 19:02:56 localhost wpa_supplicant[2159]: wlan0: Association request to the driver failed
Jan  4 19:02:56 localhost wpa_supplicant[2159]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="CARLSON5" auth_failures=1 duration=10 reason=CONN_FAILED
Jan  4 19:02:56 localhost NetworkManager[1876]: <info>  [1546628576.9638] device (wlan0): supplicant interface state: scanning -> disconnected
Jan  4 19:03:01 localhost NetworkManager[1876]: <info>  [1546628581.9478] device (wlan0): supplicant interface state: disconnected -> scanning
Jan  4 19:03:03 localhost NetworkManager[1876]: <warn>  [1546628583.4573] device (wlan0): Activation: (wifi) association took too long, failing activation
Jan  4 19:03:03 localhost NetworkManager[1876]: <info>  [1546628583.4574] device (wlan0): state change: config -> failed (reason 'ssid-not-found') [50 120 53]
Jan  4 19:03:03 localhost NetworkManager[1876]: <warn>  [1546628583.4600] device (wlan0): Activation: failed for connection 'CARLSON5'
Jan  4 19:03:03 localhost NetworkManager[1876]: <info>  [1546628583.4622] device (wlan0): state change: failed -> disconnected (reason 'none') [120 30 0]
Jan  4 19:03:03 localhost NetworkManager[1876]: <info>  [1546628583.4668] device (wlan0): supplicant interface state: scanning -> disconnected
Jan  4 19:03:03 localhost kernel: [ 1441.101113] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Jan  4 19:03:03 localhost wpa_supplicant[2159]: wlan0: Reject scan trigger since one is already pending```

kimocoder, 2019-01-10:
The floating point issue should be resolved due to use of @svpcom patch to disable "CONFIG_MP_VHT_HW_TX_MODE" which anyway is useless to our usage.

For other reports, open new ones regarding the exact issue.

closed 2019-01-10 by kimocoder

Re4son, 2019-01-23:
Compiling nicely on arm64 now. Tested with 4.14 and 4.19 kernels. Thanks guys!

kimocoder, 2019-01-23:
Great! Thanks for the info