aircrack-ng/rtl8812au #237: v5.3.4 doesn't compile on RPI (Raspbian 2018-10-09)

Yes21, 2018-11-16:
I can't compile 5.3.4 on a RPI3B :

```
pi@raspberrypi:~/rtl8812au $ make
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.14.71-v7+/build M=/home/pi/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.14.71-v7+'
Makefile:642: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.14.71-v7+'
Makefile:1649: recipe for target 'modules' failed
make: *** [modules] Error 2
```
There's no `arch/armv7l/` in Raspbian, only `arch/arm/` or `arch/arm64` ...

Yes21, 2018-11-17:
Sorry, I did a mistake ...

closed 2018-11-17 by Yes21

jbman100, 2019-02-12:
What was the mistake ? I have the same exact output ..