aircrack-ng/rtl8812au #240: No control frames in monitor mode

antifermion, 2018-12-02:
I do not receive any control frames (like Acknowledgement, RTS/CTS) in monitor mode with ALFA AWUS1900 (RTL8814AU) on 4.18.0-kali2-amd64 and 4.14.81-1-MANJARO.
This seems to be the same issue as https://github.com/aircrack-ng/rtl8812au/issues/51.
I tested the current branches v4.3.12, v5.2.20, v5.3.4 (I could not get v5.2.9 to work with the adapter).

antifermion, 2018-12-04:
I decided to perform some additional testing.
It seems the control frames are not even sent via USB.
I tested this by printing the `pkt_len` in `recvbuf2recvframe` while capturing the traffic of an Xbox One controller which permanently sends out QoS Null packets that are then ACKed by the dongle.
I also played around with the RCR flags in `hw_var_set_monitor` (rtl8814a_hal_init.c) but to no avail.
It seems like the `RCR_A*` flags might not work as described in the comments because if I only set `RCR_AMF`, no frames are received at all.
Only the flags `RCR_AAP, RCR_AM, RCR_AB` made a difference in the captured packets in my nonextensive tests.
Then again, I do not really have a clue what I am doing.

ValZapod, 2019-01-18:
@antifermion What about `iw dev <devname> set monitor flag control`

ValZapod, 2019-02-15:
@kimocoder What about command `iw dev <devname> set monitor flag control`
Han it change anything?
You have said somewhere that new drivers have more support for iw than iwconfig, right?

kimocoder, 2019-02-15:
yes, iwconfig is deprecated. I'll check it out

ValZapod, 2019-02-15:
@kimocoder So, if they deleted ioctl support than there may be a problem that `iw dev <wlan1> set type monitor` won't be enough. There are flags for type monitor (from https://wireless.wiki.kernel.org/en/users/documentation/iw#modifying_monitor_interface_flags):

> Monitor flags possible
> The following are flags you can specify:
> 
> none

> fcsfail
> plcfail
> control  //important
> otherbss
> Cook
> active
> mumimo-groupid

So if those flags are not processed by the driver, they may not work... And than oops.

kimocoder, 2019-02-15:
@ValZapod VLAN and VIF are brothers infact

/* don't support changing VLANs, you just re-create them */
	if (wdev->iftype == NL80211_IFTYPE_AP_VLAN)
		return -EOPNOTSUPP;

	switch (*mode) {
	case IW_MODE_INFRA:
		type = NL80211_IFTYPE_STATION;
		break;
	case IW_MODE_ADHOC:
		type = NL80211_IFTYPE_ADHOC;
		break;
	case IW_MODE_REPEAT:
		type = NL80211_IFTYPE_WDS;
		break;
	case IW_MODE_MONITOR:
		type = NL80211_IFTYPE_MONITOR;
		break;
	default:
		return -EINVAL;
	}

RostakaGmfun, 2019-11-05:
Does anyone have any updates on this?
Is it at least theoretically possible to receive Control frames from 8814AU?

aabedii, 2020-04-29:
I have the same problem. No control packet is captured in the monitor mode with AWUS1900 with the latest driver on Ubuntu 18.04.

srvy, 2020-10-08:
Hello,
I'm observing the same issue with TP Link Archer T9UH. Other than Block Ack Request frames I do not see any other Control frames. I'm using aircrack-ng package from git for TP link drivers. 
Can we expect the issue to be fixed? 

>uname -a
Linux kali 5.8.0-kali2-amd64 #1 SMP Debian 5.8.10-1kali1 (2020-09-22) x86_64 GNU/Linux

Thanks,
Sudhakar

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)

RostakaGmfun, 2021-05-12:
@cedricbambooza  Is it solved?

cedricbambooza, 2021-05-12:
I don't now, that's why I'm bringing it back up. 
It should get solved or closed and then reopened when something new occurs.

rodrigowue, 2022-02-07:
Hello everyone,
Any updates on this? Does the rtl8812au support the fcsfail flag?

aabedii, 2022-02-07:
+1 @rodrigowue This has been a very annoying issue for our research group. We had to purchase other equipment just because of this problem.

ajakobik, 2022-09-17:
I've the same problem with AWUS1900 (RTL8814AU).
Will it be solved any time soon?

Fz3r0, 2023-01-23:
Same issue with AWUS1900 and parrot. No ACKs or CTS/RTS..... No so much info on this huh?

lopsterman, 2023-01-23:
Sorry, but no. If you find pls tell me