aircrack-ng/rtl8812au #245: Please help for monitor mode name

CyberTheReape, 2018-12-14:
i am using tenda u12 ac1300 wifi adapter.  İ did install RTL8812AU driver and it is working now.

but  have any problem... i am putting wifi adapter into monitor mode...

but wlan1mon does not write after switching to monitor mode....wlan1 does not change when I switch to monitor mode...but monitor mode is working....I wanted to do an injection attack using the airgeddon program, but it doesn't happen.. i can deauth attack. i can catch handshake... but when the evil twin attack starts, the terminals open but not attack..deauth attack terminal says "mon: failed" ...I think I got this error because the monitor mode name is not wlan1mon...my monitor mode name "wlan1"

How do I change the name to wlan1mon when I put wifi adapter in monitor mode ??

[image]

[image]

[image]

evilginx, 2018-12-14:
İ dont understand .This is airgeddon error? Wlan1 monitor mod3 can work other evil twin attack?

kimocoder, 2018-12-14:
Yes, they should add a function to check for devices inn monitor mode, instead of looking for "wlan0mon" or "wlan1mon" or how they do it.