aircrack-ng/rtl8812au #247: Examining the connection details causes a disconnection

RainehDaze, 2018-12-22:
Looking at the connection in network manager (kde) or via the system monitor causes the connection to stop working, then disconnect entirely.

This is true for both the 5ghz and 2.6ghz bands, with separate SSIDs for each band.

francoism90, 2018-12-22:
Any logging in `dmesg` and `journalctl -xe`? Sometimes this also happens to me, and the only workaround is to reboot.

francoism90, 2018-12-22:
Would it be possible to test a different network backend? At the moment I'm using iwd and systemd-networkd, so far the connection seems to be stable.

RainehDaze, 2018-12-23:
5.2.20; someone I was talking with to get my adaptor working reported the same issue with 5.3.4.

kimocoder, 2018-12-23:
Ok. I'll check when I'm back home.

kimocoder, 2019-11-26:
This issue should be resolved. Please use latest branch, if not please re-open

closed 2019-11-26 by kimocoder