aircrack-ng/rtl8812au #249: LED Function With  RTL8811au V5.2.20

FosterSM, 2018-12-26:
I am trying to get this device's LED to function. My system is LFS/BLFS using kernel 4.19.12
I have built a vanilla module using:
git clone -b v5.2.20 https://github.com/aircrack-ng/rtl8812au.git
cd rtl8812au
make;make install
modprobe 88XXau
The driver works well on both 2.4ghz and 5ghz but I have no LED activity.
/proc/net/rtl8812au/wlan1/led_enable is set to 1.
I tried version v5.3.4 but for me this version gets a license (eventually) via  dhclient but does not pass any traffic. 
Any idea what may be wrong ?

kimocoder, 2019-02-15:
Don't know. I'm running the v5.3.4 myself, fixed the injection and txdesc buffer issue today.

gordboy, 2019-04-01:
You might have some luck with the gordboy repo at https://github.com/gordboy/rtl8812au

it has working LED control, but no monitor mode nor fancy exotic chipset support. It just does vanilla 8812au.

I actually wrote the LED support because the lights are so annoying, and I didn't want to perform LEDectomy on my dongle, nor tape it up to hide the lights ... it was never really meant to be anything other than a proof of concept.

satmandu, 2019-05-11:
@gordboy Is LED control working with your repo? ```rtw_led_enable``` doesn't seem to actually connect to anything (though you do have the CONFIG_LED_CONTROL and CONFIG_LED_ENABLE defines enabled in your Makefile).

Contrast with all the work ```rtw_led_enable``` is doing here: https://github.com/gnab/rtl8812au/pull/102/commits/b2de16a28e47b61e7f60d16820b93925e630c9e9

satmandu, 2019-05-11:
@kimocoder ```rtw_led_enable``` definitely doesn't seem to be connected to anything at all in 5.3.4.

There's a LOT of stuff surrounded by ``` CONFIG_RTW_SW_LED``` and ```CONFIG_RTW_LED``` ifdefs but I'm not seeing any of it tied to  ```rtw_led_enable```.

gordboy, 2019-06-15:
LED update. It works on my 5.2.20 repo. It doesn't work on 5.3.4 nor on 5.6.4 AFAIK. And to confuse things, Realtek changed the whole LED business so it now has policies and stuff. I am minded to remove the whole sorry mess, before it grows arms and legs.

Back to using tape over the lights ...

kimocoder, 2019-11-14:
Update to latest branch v5.6.4.2. Many issue is resolved, hopefully this one too.
If not, please re-open and let us know. Thanks!

closed 2019-11-14 by kimocoder