aircrack-ng/rtl8812au #250: Kernel panic with wiphy registration on mips (OpenWRT).

Foxtrot, 2018-12-28:
Hi,

I've been experimenting with the driver (Branch 5.2.20) on the latest snapshots of OpenWRT. It is running on mips with the 4.14.90 kernel. I've managed to successfully build the driver into an insertable kernel module but I'm running into an issue when my 8812AU is plugged in. I enabled debugging and also added some of my own logs to the driver to try and narrow down the problem, the issue seems to be (according to the stack trace) with registering the phy, I suspect that part of the wiphy struct is invalid as the dev name is null:

`[ 5715.807512] RTW: bDriverStopped:True, bSurpriseRemoved:False, bup:0, hw_init_completed:0
[ 5715.870955] RTW: iface_nums is 1
[ 5715.875145] RTW: !defined(RTW_SINGLE_WIPHY)
[ 5715.882657] RTW: Hello from rtw_cfg80211_preinit_wiphy.
[ 5715.889303] RTW: Goodbye from rtw_cfg80211_preinit_wiphy.
[ 5715.891874] RTW: rtw_wiphy_alloc((null))
[ 5715.895804] RTW: wiphy name: (null)
[ 5715.910279] RTW: rtw_wdev_alloc(padapter=c13b9000)
[ 5715.918350] RTW: rtw_os_ndev_register has arrived
[ 5715.926685] RTW: Hello from rtw_cfg80211_ndev_res_register.
[ 5715.931888] RTW: Hello from rtw_wiphy_register!
[ 5715.936434] RTW: rtw_wiphy_register((null))
[ 5715.940569] RTW: calling cfgvendor_attach on the wiphy!
[ 5715.945798] RTW: Register RTW cfg80211 vendor cmd(0x67) interface
[ 5715.960533] RTW: rtw_cfgvendor_attach returning 0
[ 5715.965238] RTW: sizeof wiphy is 4!
[ 5715.968697] ------------[ cut here ]------------
[ 5715.973450] WARNING: CPU: 0 PID: 2779 at backports-4.19.7-1/net/wireless/core.c:743 wiphy_register+0x2a4/0x9cc [cfg80211]
[ 5715.984254] Modules linked in: 88XXau(+) ath9k ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib pppoe ppp_async4
[ 5716.055447]  nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_log_common nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conn]
[ 5716.108312] CPU: 0 PID: 2779 Comm: insmod Tainted: G        W       4.14.90 #0
[ 5716.115501] Stack : 804b0000 8045dae4 00000000 00000000 8043a9b0 838359bc 83981e5c 80485307
[ 5716.123831]         80436d48 00000adb 804f3800 000002e7 c12f0000 00000001 83835970 cc3cd061
[ 5716.132147]         00000000 00000000 804f0000 00010420 00000000 00000000 00000007 00000000
[ 5716.140496]         0000451b f4e00000 0000451a 00000000 80000000 00000000 833b39f0 8338057c
[ 5716.148828]         00000009 000002e7 c12f0000 c12d6044 00000001 00000000 00000000 804f0000
[ 5716.157168]         ...
[ 5716.159579] Call Trace:
[ 5716.162024] [<8006c40c>] show_stack+0x58/0x100
[ 5716.166475] [<80082460>] __warn+0xe4/0x118
[ 5716.170526] [<80082524>] warn_slowpath_null+0x1c/0x28
[ 5716.175662] [<8338057c>] wiphy_register+0x2a4/0x9cc [cfg80211]
[ 5716.183838] [<c1235554>] rtw_wiphy_register+0xcc/0x104 [88XXau]
[ 5716.193187] [<c1235b40>] rtw_cfg80211_ndev_res_register+0x34/0xcc [88XXau]
[ 5716.203476] [<c1216118>] rtw_os_ndev_register+0xa8/0x1e0 [88XXau]
[ 5716.212991] [<c1217134>] rtw_os_ndevs_register+0xdc/0x1ac [88XXau]
[ 5716.222593] [<c1217290>] rtw_os_ndevs_init+0x28/0x50 [88XXau]
[ 5716.231759] [<c121922c>] rtw_usb_primary_adapter_init+0x7b0/0xbb0 [88XXau]
[ 5716.239636] ---[ end trace 7b2d85ec7262883d ]---`

Any input/hints are appreciated, many thanks.

kimocoder, 2019-02-15:
Take a look at the patches over here https://github.com/dl12345/rtl8812au/tree/for-15.05/patches

kimocoder, 2019-04-06:
@Foxtrot Change to the v5.3.4 branch/version, lots of fixes added by both Realtek, me, Ulli-Kroll and lwfingfer, J-forestial and others.. several openwrt fixes and it should support the openwrt tree.

v5.2.20 is going to be left behind. Open a new report if the problem persists :+1:

closed 2019-04-06 by kimocoder