aircrack-ng/rtl8812au #251: AWUS036AC vht only 20MHz

ggerla, 2018-12-30:
Hi all
I have a AWUS036AC device and I'm using hostapd to create an Access Point at 2.4GHz.
I'm not able to create an access point that is able to give me the full speed of 300Mbps.
It seems that the bandwith is always 20MHz and maximum speed is 72Mbps.
This is my hostapd conf file

interface=wlan2
driver=nl80211
country_code=IT
#ignore_broadcast_ssid=0
logger_syslog=-1
logger_syslog_level=1
#macaddr_acl=0

ssid=XXXXXXX
ieee80211d=1
ieee80211n=1
hw_mode=g
channel=6
macaddr_acl=0
auth_algs=1
#wmm_enabled=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=9
ht_capab=[HT40+][SHORT-GI-40]
vht_capab=[SHORT-GI-40][RX-STBC-4][MGN_VHT4SS_MCS9]
ignore_broadcast_ssid=0

wpa=2
wpa_passphrase=YYYYYYYYYYYYYY
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

have you any idea?

ggerla, 2018-12-30:
thanks for your answer.
even if I enabled wmm, the result is the same

Chadster766, 2018-12-30:
The line "vht_oper_centr_freq_seg0_idx" and "vht_oper_chwidth" are for 5Ghz I believe. Try removing those lines.

I think that this driver also follow the FCC 40mhz 2.4Ghz coexistence rule which mean that if another AP already has a  strong 40mhz signal at your location the drive would drop to 20mhz.

Use a free wifi analyser app to check for that scenario.

Chadster766, 2018-12-30:
I doubt you can do anything about this if I am correct that the 40mhz coexistence rule is in effect.

Chadster766, 2018-12-30:
Keep in mind the coexistence rule doesn't apply to 5Ghz.

ggerla, 2018-12-30:
ok.
there is some log to check if there is the coexistence rule in action?

Chadster766, 2018-12-30:
Try the below in your 2.4Ghz config:
```
ht_capab=[LDPC][HT40-][SHORT-GI-20][SHORT-GI-40]
```
and remove line "ht_capab=[LDPC][HT40-][SHORT-GI-20][SHORT-GI-40]"

Chadster766, 2018-12-30:
I use the below config if I want 2.4Ghz 40mhz but this is with a different wireless adapter which should make a difference as far as  hostapd is concerned:
```
interface=wlp2s0
bridge=br0
driver=nl80211
ctrl_interface=/var/run/hostapd
ssid=xxxxxxxxxx-24
ignore_broadcast_ssid=0
hw_mode=g
channel=5
auth_algs=1
wmm_enabled=1
ieee80211n=1
ht_capab=[LDPC][HT40-][SHORT-GI-20][SHORT-GI-40]
ieee80211ac=0

wpa=2
wpa_passphrase=xxxxxxxx
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
country_code=US
```

ggerla, 2018-12-30:
Strange...
which version of driver are you using?
I have this error

"Driver does not support configured HT capability [LDPC]"

probably it is because different device....

Chadster766, 2018-12-30:
Yes you can remove the "[LDPC]" I don't think it's really required.