aircrack-ng/rtl8812au #259: AWUS1900 (RTL8814AU) keep disconnecting

RnmX, 2019-01-17:
Hello,

first of all thank you very much for your hard work on aircrack-ng! I have been using it for years and it is working great.

However recently I'm facing an issue which I have been trying to resolve for a week already but with no success. I bought new WiFi dongle (AWUS1900) and I have been trying to make it work on my Kali install (in VMware). I didn't want to break my system so I did everything on newly downloaded system (the newest kali2018.4).

My problem is that when I install driver from this github repo it keeps disconnecting. I follow exactly the steps described in readme - installation via dkms and even via make, make install. As soon as I plugin the dongle into USB port, it immediately disconnects. This is what I see in dmesg:
[ 8682.062365] usb 2-2.1: new high-speed USB device number 8 using xhci_hcd
[ 8682.355028] usb 2-2.1: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[ 8682.355034] usb 2-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8682.355035] usb 2-2.1: Product: 802.11ac NIC
[ 8682.355036] usb 2-2.1: Manufacturer: Realtek
[ 8682.355037] usb 2-2.1: SerialNumber: 123456
[ 8683.528764] usb 2-2.1: USB disconnect, device number 8
[ 8755.742457] usb 2-2.1: new high-speed USB device number 9 using xhci_hcd
[ 8756.034531] usb 2-2.1: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[ 8756.034533] usb 2-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8756.034534] usb 2-2.1: Product: 802.11ac NIC
[ 8756.034535] usb 2-2.1: Manufacturer: Realtek
[ 8756.034536] usb 2-2.1: SerialNumber: 123456
[ 8757.075423] usb 2-2.1: USB disconnect, device number 9

I cannot see the device with lsusb. The dongle is not blue blinking. I tried to plug and unplug many times with no success (I tried many thing in between like resetting network manager, etc.). 

BTW I tried this on my Lenovo laptop, Surface pro and even workstation. VMware had the USB 3.0 activated. I also tried to use this adapter on windows and it works fine for normal browsing (just to confirm the card itself is not broken).

I believe I already tried pretty much everything so far without success which makes me believe there might a bug in the driver somewhere. I hope you guys could help me out,

Thank you

kimocoder, 2019-01-17:
And what did you do when you cloned this repo? Which branch did you use?

RnmX, 2019-01-17:
Hello kimocoder,

I decided to redo this whole process one more time so I can document it for you. So first step is obviously downloading VMware kali image 2018.4 fromk official kali page.

Second step is to import the vm and switch it on. Then I run "apt-get update && apt-get dist-upgrade -y" which works fine. Then I go to VM->Settings and change USB from 2 to USB 3. After that I reboot the machine.

This is uname -a after reboot: Linux kali 4.19.0-kali1-amd64 #1 SMP Debian 4.19.13-1kali1 (2019-01-03) x86_64 GNU/Linux

If I try to plug the wifi card now this is the relevant part of dmesg:
[  141.471171] usb 1-2.1: new high-speed USB device number 5 using xhci_hcd
[  141.761543] usb 1-2.1: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[  141.761550] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  141.761554] usb 1-2.1: Product: 802.11ac NIC
[  141.761557] usb 1-2.1: Manufacturer: Realtek
[  141.761561] usb 1-2.1: SerialNumber: 123456

WiFi is not working as there is no proper driver. Output of iwconfig:
lo        no wireless extensions.
eth0      no wireless extensions.

Now comes the time to install the driver, I run:
git clone -b v5.2.20 https://github.com/aircrack-ng/rtl8812au.git
cd rtl8812au

To instlall dkms:
apt-get install dkms

after that I run ./dkms-install.sh which also works fine:

About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.2.20.2/source ->
                 /usr/src/rtl8812au-5.2.20.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.19.0-kali1-amd64 KSRC=/lib/modules/4.19.0-kali1-amd64/build....................................
cleaning build area...

DKMS: build completed.

88XXau.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.0-kali1-amd64/updates/

depmod....

DKMS: install completed.
Finished running dkms install steps.

After this I reboot the system. When the system is back up I execute:
modprobe -v 88XXau with following output:

insmod /lib/modules/4.19.0-kali1-amd64/kernel/net/wireless/cfg80211.ko 
insmod /lib/modules/4.19.0-kali1-amd64/updates/88XXau.ko 

At this point I suppose all should be ready. System is freshly updated, driver installed via dkms, driver loaded via modprobe. So now comes the time to plug in WiFi card. When I do so unfortunately I get previously mentioned error:

[  129.347155] 88XXau: loading out-of-tree module taints kernel.
[  129.374537] usbcore: registered new interface driver rtl88xxau
[  273.644665] usb 1-2.1: new high-speed USB device number 5 using xhci_hcd
[  273.933685] usb 1-2.1: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[  273.933693] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  273.933699] usb 1-2.1: Product: 802.11ac NIC
[  273.933704] usb 1-2.1: Manufacturer: Realtek
[  273.933709] usb 1-2.1: SerialNumber: 123456
[  274.453349] usb 1-2.1: reset high-speed USB device number 5 using xhci_hcd
[  280.182560] usb 1-2.1: USB disconnect, device number 5

lsusb:
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 004: ID 0e0f:0008 VMware, Inc. 
Bus 001 Device 002: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

Just to make sure I restart network manager: service network-manager restart but this not help neither.

Is there more information I could provide? Or is there some step I did wrong?

Thank you!

nater1983, 2019-01-17:
Your Vendor and Product Id mite not be logged into the 88XXau driver

RnmX, 2019-01-18:
Hello ValZapod,

thank you for your feedback! Well I also thought about issue with vmware but I'm not really sure. The reason is I tried it on three different computers (Surface Pro, Lenovo laptop and workstation). Then all have different VMware version (two are on 14.x and workstation on 15.x). So I can't really see how all of them could be affected by same issue. All other devices are working great when plugged into USB they are recognised by vmware and Kali.

Another reason why I do not suspect vmware so much is that with other diver (this one https://github.com/zebulon2/rtl8814au) the card works and is all of the sudden recognised. However this driver does not support injection, etc. so it is not really usable for my purposes.

Anyway thank you very much for you effort and hopefully we can figure something out.

Thanks!

bricco1981, 2019-04-01:
it's not vm the problem, the card disconnect change the mac address and also change from 40mhz wide to 20mhz, also i cannot  make it connect on the 5ghz band at 80 mhz , only 40mhz. i have  4 pc running linux, no vm here.

kimocoder, 2019-04-07:
The 8814 chipset had got some patches, so try "git pull" and build/install it again and see if the problem is gone.

Orphan-Crippler, 2019-09-10:
I have been experiencing the same problem with kali 2019.3 using VM Ware, It is installing the right version 5.2.20, i double checked since I am using apt-get to install. I did however get the adapter to work somewhat by using a regular micro USB cable. It seems to work just fine i can put it in monitor mode and injection is working. It does not use the 5Ghz band though and im sure it is lacking in speed from not using the USB 3.0 cable.

kimocoder, 2019-11-14:
Update to the latest branch v5.6.4.2, hopefully the issue is resolved. If it's not, feel free to open this issue report again. Thanks!

closed 2019-11-14 by kimocoder