aircrack-ng/rtl8812au #261: ASUS USB AC68 soft APslower with 5.2.20

logan001, 2019-01-25:
I've been using these drivers for some time now for soft AP. It seems that your official realease is much slower than 5.3.4 release. the phone (android) connects on 5.2.20 with 135 Mbps and speed test on my internet provider server reach 40Mbps while on 5.3.4 my phone connects with 200Mbps and download goes to 100-110 Mbps. Just so to be clear computer reach the same server with 950-1000Mbps so its not of problem of internet.

I think that 5.3.4 has some issues with aircrack suite but i'm not sure. i was never able to capture handshake with it. i can do it with 5.2.20 though.

my hostapd.conf is
`interface=wlx2c4d5406dbfb
#country_code=
driver=nl80211
ssid=wifi
hw_mode=a
channel=44
ieee80211n=1
ieee80211ac=1
wmm_enabled=1
ieee80211d=0
ieee80211h=0

ht_capab=[RX LDPC][HT40+][HT40-][MAX-AMSDU-7935][DSSS_CCK-40][RX-STBC3][SHORT-GI-40][RX-STBC][RXLDPC][DSSS_CCK-40]
vht_capab=[Max-MPDU-11454][SHORT-GI-80][RX-LDPC][TX-STBC][TX-STBC-2BY1][SU-BEAMFORMEE][HTC-VHT][SU-Beamformer]
wpa=2
wpa_passphrase=1234567890
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

`
P.S.: Does anybody knows if this device caped at 200Mbps for soft ap? or can i make it go higher? i have same speed with windows soft ap.

ValZapod, 2019-01-26:
@logan001 Did you read this? https://github.com/aircrack-ng/rtl8812au/issues/253
80 MHz is not working on 5.2.20, so... 135 Mbps  is 40 MHz channel as you can read there https://en.wikipedia.org/wiki/IEEE_802.11ac#Data_rates_and_speed

> my phone connects with 200Mbps

And what is the actual number? Maybe 260 or 234? It is important because it is usually unique.

logan001, 2019-01-26:
i've read a bit through that one but i thought that that person is trying to connect to router and since i'm creating a soft AP, i thought its something different.
What do you mean by "260 or 234"? where/how can i find this info?

ValZapod, 2019-01-26:
@logan001 And where did you find the 200 Mbit/s value? I need not rounded value (if it was rounded, of cause).

ValZapod, 2019-01-26:
/facepalm Your phone does not support 802.11ac at all! Or actually cpu Intel Atom Z3580
Only 802.11n, so no 80 Mhz. And i suppose only 2 SS... So up to 300 mbit/s.
Or maybe you have Zenfone 2 Deluxe?

logan001, 2019-01-26:
are you sure it does not support AC? according to asus it does. Windows drivers set to only AC and my phone still can connect to it. i have no other way to check this. And speed from windows soft ap is the same. so it might be a limitation of my phone too.

ValZapod, 2019-01-26:
Ok, it supports 802.11ac. And it does support 80MHz channel bandwidth. But it support only 1 Spacial Stream! So the problem is in #253, actually. Again look at the table https://en.wikipedia.org/wiki/IEEE_802.11ac#Data_rates_and_speed

kimocoder, 2019-02-15:
hostapd and wpa_cli isn't the most supported by these drivers, but.. I've added tons of documentation in the folder with the same name about it.

Besides that, today I've fixed the sluggish injection on v5.3.4, I also got a fix for "reaver"'s pcap_activate error thanks to heads up by @kcdtv I actually run it smooth now. And yes, it's a little quicker than the v5.2.20(.2) and a notch more stable too.

We'll move over to v5.3.4 soon.. still got some mediums to get rid of.

closed 2019-02-15 by kimocoder